openscad/openscad #4194: HIDAPI not picked up by macOS build

t-paul, 2022-03-27:
CircleCI homebrew build has the dependency built:
```
hidapi 0.11.0 already installed - not building
```

but cmake is not picking it up:
```
-- HIDAPI: disabled
...
-- - HID API Driver (disabled)
```

thehans, 2022-03-28:
It looks like hidapi [uses `pkg-config` ](https://github.com/libusb/hidapi/tree/master/pc) 

However our CMakeLists.txt is checking with `find_package`, 
https://github.com/openscad/openscad/blob/d444a31b8674d786eec16a181c26989d4949fb35/CMakeLists.txt#L348-L349
so it goes through our own project's [`/cmake/Modules/FindHidAPI.cmake`](https://github.com/openscad/openscad/blob/master/cmake/Modules/FindHidAPI.cmake)

When looking at that file, it does indeed use cmake's `pkg-config`-based function `PKG_CHECK_MODULES` to check specifically for `hidapi-libusb`

But [as the documentation shows](https://github.com/libusb/hidapi#hidapi-has-four-back-ends), libusb backend is NOT used on Mac.  So i think it needs to check for just plain `hidapi` with no suffix?  

For reference on Linux, I found the pkg-config files like so:
```
$ sudo find / -iname "*hidapi*.pc"
/usr/lib/x86_64-linux-gnu/pkgconfig/hidapi-hidraw.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/hidapi-libusb.pc
```
So there's no plain `hidapi.pc` for linux, but I suspect that is what it is named on mac.  We will likely need a way to check for both `hidapi` and `hidapi-libusb` in case one fails.

I think the cleanest way to do this would be to use `pkg_search_module` instead of `pkg_check_modules`:
https://cmake.org/cmake/help/latest/module/FindPkgConfig.html#command:pkg_search_module
> The behavior of this command is the same as [pkg_check_modules()](https://cmake.org/cmake/help/latest/module/FindPkgConfig.html#command:pkg_check_modules), except that rather than checking for all the specified modules, it searches for just the first successful match.

So the line at top of our FindHidAPI.cmake should become:
```
pkg_search_module(PC_HIDAPI QUIET hidapi hidapi-libusb)
```

thehans, 2022-03-28:
Additionally, I think we should change the message here to be a `FATAL_ERROR`, since its within a conditional that `ENABLE_HIDAPI` was `ON`, and it failed to enable it:
https://github.com/openscad/openscad/blob/d444a31b8674d786eec16a181c26989d4949fb35/CMakeLists.txt#L365

t-paul, 2022-03-29:
Ah, nice, thanks, maybe I can try that via the CI.

Hmm, not sure about the disabled part. But I suppose you are right. The default is `ON` so it would fail if it's not available, but there's still the option to specifically set it to `OFF`. There's a couple of more cases with the same logic, so maybe those should be unified.

commit f50aa883 (2022-12-27): Make hidapi findable on macOS. Fixes #4194 (#4472)
closed 2022-12-27 by kintel