openscad/openscad #4199: [Feature request] Interrupt preview

nesdnuma, 2022-04-03:
Hello,

Quite often the preview appear to reveal an issue in the code very quickly, possibly with a lot of error messages appearing in the console. However the process can be quite long but is executed until the end, and cannot be interrupted whereas a normal rendering can. Could it be possible to add this possibility to interrupt the preview?

MichaelPFrey, 2022-04-03:
> Quite often the preview appear to reveal an issue in the code very quickly, possibly with a lot of error messages appearing in the console. 

Are your talking about warnings or errors?

If you talk about warnings: please enable "stop on first warning" in the preferences and see if that helps.

If you talk about actual errors: Do you have sample code?

> However the process can be quite long but is executed until the end, and cannot be interrupted whereas a normal rendering can.

OpenSCAD has two/multiple phases: Interpreting/evaluating the script and then creating the preview/render.

If you provide sample code, we can look into where it happens.

Assuming it is actual errors (not warnings) a user level interrupt request is unlikely to help, as the error handling is mainly creating the error message, trace and clean up - things that would also makes sense during a user interrupt.

t-paul, 2022-04-03:
Duplicate of #1195.

closed 2022-04-03 by t-paul

nesdnuma, 2022-04-03:
Ah yes, "Stop on first warning" is "Arrêter après la première alerte" ("Avancé") in the french localization. This will help a lot. Thanks!