openscad/openscad #4201: Add official 3DMouse  support

palapalal, 2022-04-05:
Hi guys,
I am currently integrating official support for 3Dconnexion devices in OpenSCAD, including 3D navigation and commands export.
In our navigation modes, we make use of hit testing in order to recompute the rotation centre during the 3D navigation. Since there is no hit-testing function already implemented in OpenSCAD, I'm trying to use the Z-buffer for computing the distance between the hit testing origin and the first intersection of the hit test ray on the displayed mode

After few trials, I could get a correct solution for getting the distance between the ray of the camera and the model.
A correct solution is given by rendering everytime the model, but this operation is very time consuming.
Do you have any suggestion for that?

Here is the sample
```c++
	std::function<void(const Renderer::shaderinfo_t *shaderInfo)> prepareDrawer =[win](const Renderer::shaderinfo_t *shaderInfo){
		auto r = win->qglview->getRenderer();
		if(r){
    		r->prepare(true, false, shaderInfo);
		}
	};
	std::function<Eigen::Vector2d()> mousePosProvider = [win]{
		QPoint pos = QCursor::pos();
		pos = win->qglview->mapFromGlobal(pos);
		return Eigen::Vector2d({pos.x(),pos.y()});
	};
	std::function<void(const Renderer::shaderinfo_t *shaderInfo)> drawer =[win](const Renderer::shaderinfo_t *shaderInfo){
		auto r =  win->qglview->getRenderer();
		if(r){
    		r->draw(true, false, shaderInfo);
		}

double GetZBufferDepth(Eigen::Vector3d const &position, Eigen::Vector3d const &direction,
											 double diameter, Camera const &cam,
											 std::function<void(const Renderer::shaderinfo_t *)> prepareDrawer,
											 std::function<void(const Renderer::shaderinfo_t *)> drawer)
{
	Renderer::shaderinfo_t shaderInfo = CreateShaderInfo("Preview.vert", "Preview.frag");
	prepareDrawer(&shaderInfo);

	QOpenGLFramebufferObjectFormat fboFormat;
	fboFormat.setSamples(0);	
	fboFormat.setAttachment(QOpenGLFramebufferObject::Depth);
	
	std::unique_ptr<QOpenGLFramebufferObject> framebuffer;
	framebuffer.reset(new QOpenGLFramebufferObject(cam.pixel_width, cam.pixel_height,
		fboFormat));
	framebuffer->release();
	framebuffer->bind();

	glViewport(0, 0, cam.pixel_width, cam.pixel_height);

	GLint viewport[4];
	glGetIntegerv(GL_VIEWPORT, viewport);
	auto aperture = ComputeAperture(viewport, diameter, cam.getViewVolume());
	// position the pick point in the middle of the viewport
	Eigen::Vector2d pickPoint(static_cast<int>((viewport[2] + 1) / 2), static_cast<int>((viewport[3] + 1) / 2));

	auto far_value=1.0f;
	glClearBufferfv(GL_DEPTH, 0, &far_value);
	GlPickInit(cam, pickPoint, aperture);

	GlModelView (cam);

	GLdouble modelMatrix[16];
	glGetDoublev(GL_MODELVIEW_MATRIX, modelMatrix);
	GLdouble projectionMatrix[16];
	glGetDoublev(GL_PROJECTION_MATRIX, projectionMatrix);

	drawer(&shaderInfo);

	glFlush();
	glFinish();

	Eigen::Vector2d readPoint(
			{pickPoint.x() - (aperture.x() / 2.0), pickPoint.y() - (aperture.y() / 2.0)});

	if(aperture.x() <0 || aperture.y()<0){
		ResetOpenGl();
		framebuffer->release();
		return cam.getViewVolume().farVal;
	}
	const long bufferSize = static_cast<double>(aperture.x()) * static_cast<double>(aperture.y());
	if(bufferSize<0){
		ResetOpenGl();
		framebuffer->release();
		return cam.getViewVolume().farVal;
	}
	std::unique_ptr<GLfloat[]> depthBuffer(new GLfloat[bufferSize]);

	glReadPixels(readPoint.x(), readPoint.y(), aperture.x(), aperture.y(), GL_DEPTH_COMPONENT,
							 GL_FLOAT, depthBuffer.get());

	ResetOpenGl();
	framebuffer->release();

	int i, pos = 0;
	GLfloat depth = depthBuffer[pos];

	for (i = 1; i < bufferSize; ++i) {
		if (depthBuffer[i] < depth) {
			pos = i;
			depth = depthBuffer[pos];
		}
	}
	if (depth == 1.) {
		return cam.getViewVolume().farVal;
	}

	size_t y = pos / aperture.x();
	size_t x = pos - y * aperture.x();
	GLdouble objx, objy, objz;
	gluUnProject(static_cast<GLdouble>(readPoint.x() + x), static_cast<GLdouble>(readPoint.y() + y),
							 static_cast<GLdouble>(depth), modelMatrix, projectionMatrix, viewport,
							 &objx, &objy, &objz);
	Eigen::Vector3d hit(
			{static_cast<double>(objx), static_cast<double>(objy), static_cast<double>(objz)});

	return (hit - position).dot(direction);
}
```

Thank you in advance

palapalal, 2022-04-06:
The integration is done using the official 3Dconnexion SDK v. 4.0. It will improve the user experience in Windows and MacOS by offering advanced 3D navigation. It is not needed to statically include proprietary libraries into the project, as they can be dynamically loaded and available to customers that have the 3Dconnexion driver installed.

This new integration can be disabled in CMake if not desired. For the Linux systems, the current 3D mouse support of OpenSCAD will be kept.

t-paul, 2022-04-06:
What about the licence question? My earlier look at the SDK license showed that it is very much incompatible with open source (e.g. see discussion in https://github.com/openscad/openscad/issues/4188#issuecomment-1079488726). Has that changed?

It would be much appreciated to see the official support, but OpenSCAD uses a large number of libraries, some of which are GPL so we need to make sure there's no incompatibilities.

OpenSCAD  code itself is GPL2+ which seems to be a bit unclear in regard to dynamic linking but the OpenSCAD binary is effectively GPL3 at this point (due to libCGAL) and that seems to be quite clear (in my non-lawyer understanding) that dynamic linking is not a way to workaround licence conditions.

palapalal, 2022-04-06:
If the dynamic linking is not an option, we can propose an approach based on web socket. This approach is normally used in web applications: the 3Dconnexion driver installs a server component that listens for a client application. The client application exchanges the 3D viewport data with the server, and it receives back the new camera position according to the user input.
The needed dependencies in this case are boost (for the web socket) and an open-source JSON library (nlohmann), no need to statically or dynamically link closed-source libraries. Would this approach be compliant with your license?

t-paul, 2022-04-06:
I don't see any issues with that approach. Calling external service interfaces should be safe in my understanding.
Those 2 dependencies already exist, `boost` is checked for version 1.36 via `cmake`, but I would assume it's actually more like >= 1.56 as listed in the readme. JSON `nlohmann` we have in `src/ext/json/`, currently at 3.10.4.

jordanbrown0, 2022-04-06:
Dynamic linking should be an adequate "barrier" between GPL and proprietary components.  On many systems, **everything** ends up dynamically linked against proprietary components.  (On Solaris, where I work, it's libc and numerous other libraries.  On Windows, it's a number of DLLs.)

Even header files are probably OK.  On many systems, programs are required to build against proprietary header files.

What I would worry about are the license conditions on the SDK itself.  You are allowed to distribute the SDK **in object code form**.  Autodesk can build AutoCAD and distribute it, no problem.  But OpenSCAD needs to allow people to build from source, and needs to have automatic builds on from publicly accessible sources, and the SDK license does **not** authorizing redistributing the SDK in source form.  Anybody building against it would need to individually agree to the SDK license.  Further, the license requires you to maintain the trade secret materials in the SDK... which would include, say, the header files.

Theoretically, Torsten could download the SDK, agree to its terms, build OpenSCAD against it in a private environment, and distribute the resulting binaries - but the automated builds couldn't, and any others would need to separately download the SDK, separately agree to its terms, and build in a private environment.

Does 3DCONNEXION really intend to block open-source use of their SDK?  Maybe, maybe not, but unfortunately I have to agree with Torsten that it's legally problematic.

t-paul, 2022-04-07:
I'm not sure this is possible with the normal preview renderer. It uses the Z-buffer in a special way. The special render pass used for the jump-to-source feature might help. @Tomatower do you have some input on that? Could the Z-buffer of the mouse-selection-renderer be used for distance calculation too?

palapalal, 2022-04-22:
> I'm not sure this is possible with the normal preview renderer. It uses the Z-buffer in a special way. The special render pass used for the jump-to-source feature might help. @Tomatower do you have [...]

If this feature consist in using preview vertex and shader files. I tried but didn't got good results (the code is the one shown in the fist message). I might also forget or miss used something .

Tomatower, 2022-04-22:
The selection renderer uses its own framebuffer, not interfering with the F5-Buffer.

The logic from the selection buffer can be used to query for the depth value, check https://github.com/openscad/openscad/blob/d62b8c0cc4ac952b8edc0d9b8936da58006c8483/src/gui/MouseSelector.cc#L188 with ` 

The selection rendering disables/replaces a lot of the shaders, and it could be improved speed-wise using the VBO rendering, and keeping the VBOs in GDDR.

rhandel, 2022-09-17:
Two questions:
1) Any updates?
2) Need a tester?

RAH

rhandel, 2022-11-12:
> Hi Rhandel, We are sorry for the delay, I will soon create a merge request. Best

Palapalal,

Is there still anyone integrating the SDK with OpenSCAD?  Blender, a MAJOR open source software package, uses the 3D Mouse SDK outright AFAIK and has no issues.  I just wonder why no one has already integrated this into OpenSCAD.

Thank you,

RAH

rcarmo, 2024-05-28:
I've been digging into this due to #5153, and while also [digging through the Blender source code](https://github.com/blender/blender/blob/ceeaea0c8a94f1ebe0f695d61befeb046ad35697/intern/ghost/intern/GHOST_NDOFManagerUnix.cc#L91) I found [a dependency to `libspnav`](https://github.com/blender/blender/blob/ceeaea0c8a94f1ebe0f695d61befeb046ad35697/build_files/build_environment/cmake/versions.cmake#L400), which on its [README](https://github.com/FreeSpacenav/libspnav/blob/master/README.md) says:

> Also, libspnav provides a magellan API wrapper on top of the libspnav API, which allows existing applications written for the official SDK to be transparently recompiled against libspnav and be free of the 3Dconnexion licensing conditions.

Blender works perfectly with the new Bluetooth SpaceMouse, too, so I am assuming it is going through [this handler](https://github.com/blender/blender/blob/ceeaea0c8a94f1ebe0f695d61befeb046ad35697/intern/ghost/intern/GHOST_NDOFManagerUnix.cc#L42) and using this somehow, although I cannot figure out how it gets built into the macOS binary just yet.

rhandel, 2024-10-09:
@rcarmo Add that Prusa Slicer also open source (at some level), also integrates the 3D Mouse.
I have to believe that there are special circumstances where this is made possible.

Thank you,

RAH

BenBergman, 2025-05-20:
FreeCAD also integrates with the official 3Dconnexion driver on macOS. It looks like it was recently redone by a 3Dconnexion employee in [this PR](https://github.com/FreeCAD/FreeCAD/pull/12929) to use their libnav interface (I think it is a websocket inteface also used by web CAD tools like Onshape?).

@pskowronskiTDx since you implemented that PR, can you provide any insight into licensing concerns above?

t-paul, 2025-05-20:
There is still that:

https://github.com/FreeCAD/FreeCAD/pull/12929/files?diff=unified&w=0#diff-3ccd83eea8150cfb8df73e6f92d2ed9596c8bdfb79c01663b322313cb4a297c7R43-R48

minglecm, 2025-11-10:
> There is still that:
> 
> https://github.com/FreeCAD/FreeCAD/pull/12929/files?diff=unified&w=0#diff-3ccd83eea8150cfb8df73e6f92d2ed9596c8bdfb79c01663b322313cb4a297c7R43-R48

@t-paul would you be open to a solution that requires a build flag to enable official 3Dconnexion support?  I.e. the integration code would be in the tree but would only be dynamically linked against `TDxNavLib.lib` by the end user when using the cmake build flag? I think that meets the requirements of the individual licenses with the only limitation being an official binary couldn't be distributed with that library linked.  I think this is how some other GPL projects have dealt with proprietary GPU drivers/codecs.

I know a build flag is certainly less than ideal but it seems like it would at least be a path to official 3D mouse support on Windows/macOS.

Alternatively, I think it's possible to discover the library at runtime if the user has it installed without dynamically linking to it at build time.  E.g. `dlopen`.

t-paul, 2025-11-10:
> [@t-paul](https://github.com/t-paul) would you be open to a solution that adds optional 3Dconnexion SpaceMouse support behind a CMake build flag?

No, I have no plan on discussing that over and over again when someone finds the flag and the fact it's not enabled in the official builds.

As was discussed earlier, there seems to be a technical solution that does not have the license issues. That makes it even more questionable going for the problematic solution.

> Alternatively, it might be possible to load the library at runtime (e.g. via `dlopen`/`LoadLibrary`) if present on the system, so the feature would activate automatically when available. That would avoid any build-time dependency and allow the official binaries to remain completely GPL-compliant.

In my view that is NOT GPL compliant at all unless someone brings a lawyer explicitly stating it's fine. I'm pretty sure that's not going to happen as that is exactly the same situation as with the nvidia kernel driver and as far as I know that never ended up declaring dynamic loading/linking is a loop hole to get around the license restrictions.

dewi-ny-je, 2026-10-11:
If you bind to official SDK/drivers you are going to cut support to old devices (which are very popoular in 2nd hand market).
Maybe mention that there are two open source projects to support those: the original Mac project https://github.com/consi/axial
 and my Windows port https://github.com/dewi-ny-je/spacepilot-windows 
Neither one is GPL, but they are free (surely free from planned obsolescence)

rcarmo, 2026-10-11:
It's been a while since I used OpenSCAD at my main desk, but I might look into this later.