openscad/openscad #4202: Error compling for Ubuntu 20.04.2

Ruan-hub, 2022-04-06:
Hello everyone.
     I have encountered the ambiguous problem while compiling the code. The error came from the src/openscad.cc file. The Following is the error message. The boost version in my system is 1.71. Do you have any ideas about the problem? Thanks in advance.
`src/openscad.cc: In function ‘void set_render_color_scheme(std::string, bool)’:
src/openscad.cc:310:100: error: call of overloaded ‘join(std::__cxx11::list<std::__cxx11::basic_string<char> >, const char [2])’ is ambiguous
  310 |   LOG(message_group::None,Location::NONE,"",(boost::join(ColorMap::inst()->colorSchemeNames(), "\n")));
      |                                                                                                    ^
In file included from /usr/include/boost/algorithm/string.hpp:24,
                 from src/printutils.h:7,
                 from src/exceptions.h:6,
                 from src/feature.h:9,
                 from src/module.h:5,
                 from src/FileModule.h:8,
                 from src/comment.h:3,
                 from src/openscad.cc:28:
/usr/include/boost/algorithm/string/join.hpp:46:9: note: candidate: ‘typename boost::range_value<T>::type boost::algorithm::join(const SequenceSequenceT&, const Range1T&) [with SequenceSequenceT = std::__cxx11::list<std::__cxx11::basic_string<char> >; Range1T = char [2]; typename boost::range_value<T>::type = std::__cxx11::basic_string<char>]’
   46 |         join(
      |         ^~~~
In file included from /usr/local/include/CGAL/Boolean_set_operations_2/Polygon_conversions.h:16,
                 from /usr/local/include/CGAL/Boolean_set_operations_2/Bso_internal_functions.h:24,
                 from /usr/local/include/CGAL/Boolean_set_operations_2/complement.h:34,
                 from /usr/local/include/CGAL/Boolean_set_operations_2.h:21,
                 from /usr/local/include/CGAL/Minkowski_sum_2/Minkowski_sum_decomp_2.h:21,
                 from /usr/local/include/CGAL/minkowski_sum_2.h:24,
                 from src/cgal.h:23,
                 from src/CGAL_Nef_polyhedron.h:4,
                 from src/openscad.cc:57:
/usr/include/boost/range/join.hpp:66:1: note: candidate: ‘boost::range::joined_range<const SinglePassRange1, const SinglePassRange2> boost::range::join(const SinglePassRange1&, const SinglePassRange2&) [with SinglePassRange1 = std::__cxx11::list<std::__cxx11::basic_string<char> >; SinglePassRange2 = char [2]]’
   66 | join(const SinglePassRange1& r1, const SinglePassRange2& r2)
      | ^~~~
src/openscad.cc: In function ‘int main(int, char**)’:
src/openscad.cc:967:106: error: call of overloaded ‘join(const std::vector<std::__cxx11::basic_string<char> >, const char [4])’ is ambiguous
  967 |   ("view", po::value<CommaSeparatedVector>(), ("=view options: " + boost::join(viewOptions.names(), " | ")).c_str())
      |                                                                                                          ^
In file included from /usr/include/boost/algorithm/string.hpp:24,
                 from src/printutils.h:7,
                 from src/exceptions.h:6,
                 from src/feature.h:9,
                 from src/module.h:5,
                 from src/FileModule.h:8,
                 from src/comment.h:3,
                 from src/openscad.cc:28:
/usr/include/boost/algorithm/string/join.hpp:46:9: note: candidate: ‘typename boost::range_value<T>::type boost::algorithm::join(const SequenceSequenceT&, const Range1T&) [with SequenceSequenceT = std::vector<std::__cxx11::basic_string<char> >; Range1T = char [4]; typename boost::range_value<T>::type = std::__cxx11::basic_string<char>]’
   46 |         join(
      |         ^~~~
In file included from /usr/local/include/CGAL/Boolean_set_operations_2/Polygon_conversions.h:16,
                 from /usr/local/include/CGAL/Boolean_set_operations_2/Bso_internal_functions.h:24,
                 from /usr/local/include/CGAL/Boolean_set_operations_2/complement.h:34,
                 from /usr/local/include/CGAL/Boolean_set_operations_2.h:21,
                 from /usr/local/include/CGAL/Minkowski_sum_2/Minkowski_sum_decomp_2.h:21,
                 from /usr/local/include/CGAL/minkowski_sum_2.h:24,
                 from src/cgal.h:23,
                 from src/CGAL_Nef_polyhedron.h:4,
                 from src/openscad.cc:57:
/usr/include/boost/range/join.hpp:76:1: note: candidate: ‘boost::range::joined_range<SinglePassRange1, SinglePassRange2> boost::range::join(SinglePassRange1&, SinglePassRange2&) [with SinglePassRange1 = const std::vector<std::__cxx11::basic_string<char> >; SinglePassRange2 = const char [4]]’
   76 | join(SinglePassRange1& r1, SinglePassRange2& r2)
      | ^~~~
/usr/include/boost/range/join.hpp:66:1: note: candidate: ‘boost::range::joined_range<const SinglePassRange1, const SinglePassRange2> boost::range::join(const SinglePassRange1&, const SinglePassRange2&) [with SinglePassRange1 = std::vector<std::__cxx11::basic_string<char> >; SinglePassRange2 = char [4]]’
   66 | join(const SinglePassRange1& r1, const SinglePassRange2& r2)
      | ^~~~
make: *** [Makefile:4475: objects/src/openscad.o] Error 1
`

t-paul, 2022-04-06:
Compiling which specific version?

Ruan-hub, 2022-04-06:
It was openscad-2021.01

t-paul, 2022-04-06:
Right, that does indeed conflict, you can
* use https://github.com/openscad/openscad/commits/2020.12-branch which has some patches including one that should fix this build error
* use `master` branch

Depends on why you are builing from source.

Ruan-hub, 2022-04-06:
Thank you for your reply. I will try the master branch.

closed 2022-05-07 by t-paul