openscad/openscad #4203: Feature request: Associative arays

lucky62, 2022-04-06:
Hello friends,

I am thinking why this very good software has not the **associative arrays**... (sorry vectors..)
I found a lot of workarounds using some kind of search functions, e.g.: [here](https://www.thingiverse.com/groups/openscad/forums/general/topic:5610) or [here](https://stackoverflow.com/questions/55448011/associative-arrays-in-openscad).

What is currently working is that first three items from the array can be accessed by x,y,z indexes like:
```
array.x
array.y
array.z
```
So why we cannot access the any item by similar way?

I can imagine that the array can be specified by the following example (or something similar):
```
array1 = [
   index0: "value0",
   index1: [ "value1-0", 1.1 ]
   index2: [ 
      index2-0: "value2-0",
      index2-1: "value2-1",
   ]
]
```
Then items might be accessible by normal numerical indexes or by "named indexes"
So these two/three variants will be equivalent:
```
array1[0]
array1.index0
array1["index0"]    // probably better variant?...
```
And some nested arrays samples:
```
array1[1][1]
array1.index1[1]   // nested array has only numeric indexes
array1.index2.index2-1
array1.index2["index2-1"]    // combined way
array1["index2"]["index2-1"]    // combined way
array1[2]["index2-1"]    // combined way
array1["index2"][1]    // combined way
```

The array can be stored in memory by the same way as now with additional info which will allow to convert the "named indexes" to the numeric indexes - then Internally the item will be retrieved by numeric index as usually.

This can be very useful feature. What do you think?...

jordanbrown0, 2022-04-06:
Some of the support is there.  Internal functions (notably textmetrics() and fontmetrics(), ref #3684 ) can create objects with strings as their keys, and most of the operations that you might expect can be done on them.  You can also import JSON files to create objects, ref #3891 .

What you **can't** do is to create them directly from an OpenSCAD program.  Ref #3088 .

JustinSDK, 2022-04-07:
I implement  my own [`hashmap`](https://openhome.cc/eGossip/OpenSCAD/lib3x-hashmap.html) in my [library](https://github.com/JustinSDK/dotSCAD). You may take it a try.

lucky62, 2022-04-07:
@jordanbrown0, JSON objects are very useful.
Are you (or someone else) planning to allow to create a JSON in OpenSCAD program?... :-)

@JustinSDK, hashmap can help also, but still it is a workaround to simulate missing feature..

jordanbrown0, 2022-04-07:
I'll do it eventually, if somebody else doesn't get to it first.

rjcarlson49, 2022-08-09:
BOSL2 has a structs library that implements associative arrays like this. The syntax vector["keyname"] could easily be made to be equivalent to the BOSL2 struct equivalent, with one exception. In BOSL2 struct_val(), there is a default argument that is useful. It defaults to undef. I prefer this to a .key syntax.

jordanbrown0, 2022-08-10:
OpenSCAD objects (discussed above) can be accessed using either obj.key or obj["key"] syntax.  They are like JavaScript objects in that way.

Jasonkoolman, 2025-01-04:
+1 would love to see this - `textmetrics()` has a similar implementation as it returns an object (literal).

jordanbrown0, 2025-01-04:
PR#4478 has this functionality, but is waiting on review.

closed 2026-05-07 by jordanbrown0