openscad/openscad #4204: OpenSCAD 2021.01 renders imported 3MF files weirdly

hroncok, 2022-04-07:
I don't know if this is some problem in our Fedora packaging of lib3mf or openscad, but I'd like to get it fixed and I have no idea how. Note that I cannot reproduce this with Debian/Ubuntu OpenSCAD package, just in Fedora. My openscad and lib3mf are installed from the Fedora repos, I run Fedora 35, x86_64.

```
openscad-2021.01-4.fc35.x86_64
lib3mf-2.0.1-2.fc35.x86_64
```

This also happens with `lib3mf-2.2.0-1.fc35.x86_64` from [my own Fedora pull request](https://src.fedoraproject.org/rpms/lib3mf/pull-request/2), no magic fix happened by upgrading.

## Steps to reproduce

 1. Open an OpenSCAD model, e.g. from examples. I used the basic CSG example.
 2. Render and export as 3MF.
 3. In another OpenSCAD file, import the 3MF file, e.g. `import("CSG.3mf");` and render/preview it.

## Expected result

The render/preview should look like the original model.

[image]

## Actual result

The render/preview looks weird.

[image]

This looks like a problem with import, rather than export, because the exported 3MF file looks normal in PrusaSlicer.

[image]

I would appreciate any pointers. I've searched 3MF in issues here, but could not find anything. Thanks

nophead, 2022-04-07:
Looks like the points get snapped to a very coarse grid, possibly just integers. Something wrong with text to float conversion.

t-paul, 2022-04-07:
Locale issue https://github.com/3MFConsortium/lib3mf/issues/282

There's no full support for lib3mf v2.x at this point, but I'm pretty sure it's not an OpenSCAD issue.

hroncok, 2022-04-07:
Oh, the infamous [scanf breaks with Qt apps and locales with decimal commas](https://github.com/WoLpH/numpy-stl/issues/52#issuecomment-440077501) :facepalm: 

My Fedora system is Czech and the Ubuntu/Debian system I tested this on is English.

Feel free to close this here, if you are not interested in a workaround.

hroncok, 2022-04-07:
The workaround would be to wrap lib3mf library calls with temporary numeric locale reset to C.

t-paul, 2022-04-07:
Lets keep it open to monitor what's going on. I'll label it as discussion.

I hope at some point we can fully move to v2.x. so it might be usefull to keep the reference.

t-paul, 2023-08-25:
Upstream Patch submitted - https://github.com/3MFConsortium/lib3mf/pull/322.

makeasnek, 2023-08-25:
> Want to back this issue? **[Post a bounty on it!](https://app.bountysource.com/issues/112040451-openscad-2021-01-renders-imported-3mf-files-weirdly-due-to-locale-lib3mf-bug?utm_campaign=plugin&utm_content=tracker%2F52063&utm_medium=issues&utm_source=github)** We accept bounties via [Bountysource](https://app.bountysource.com/?utm_campaign=plugin&utm_content=tracker%2F52063&utm_medium=issues&utm_source=github).

     

Please do not use bountysource. Many devs have had trouble getting paid there. You can check out this lemmy community as an alternative https://lemmy.ml/c/bugbounties

For statements from devs who have been unable to cash out from bountysource see:
https://github.com/bountysource/core/issues

t-paul, 2023-08-25:
@makeasnek feel free to post that as a separate issue, it's not really useful having that as comment on some random issue which not even has a bounty. I'll mark both comments as off-topic here.