openscad/openscad #4206: can't iterate variables in loops?

Antiswitcher, 2022-04-07:
```
tempointp = [10, 0, 0];
templength = 100;
tempgirth = 20;
temporarypoint = 0;
// point is xyz + xyz rotation

//    translate([0 - (sqrt(2)* tempgirth)/2 , 0, 0])

//        rotate(a=45, v=[0,1,0])

// rotate a is angle, v is axis, xyz

for (a = [1:1:5]){
    tempointp = [tempointp[0] + (10*a) , 0 , 0];
    temporarypoint = temporarypoint + 1;
    
    echo(temporarypoint)
        translate([tempointp.x, 0, 0])
    rotate(a=45, v=[0,0,1])
                cube([tempgirth, templength, tempgirth]);
    }
```
    when running the above code, echo reports "1" on each loop. How am I meant to change a variable within a loop that's a separate from **a**?

jordanbrown0, 2022-04-07:
You can't.  OpenSCAD variables are really more like named constants than like variables; each iteration is getting you a new instance of "temporarypoint" that is then discarded at the end of the iteration.

Depending on the exact need, some options are:
- recursion
- list comprehension
- C-style "for" loops

thehans, 2022-04-07:
Please keep discussions / introductory questions about the language to the mailing list or IRC.

closed 2022-04-07 by thehans