openscad/openscad #4217: Error installing -- "bad variable name" when run uni-build-dependencies.sh

lakfel, 2022-04-14:
Hi,
Tried to install Openscad. When checking the dependencies `./scripts/check-dependencies.sh`  One dependency (CGAL) has a version below the minimum. Following the instructions, I tried to run  `./scripts/uni-build-dependencies.sh` but I does not run and shows me always the same message

[image]

Not sure if I have missed something

t-paul, 2022-04-14:
Version 5.4 of CGAL is strongly recommended as this provides some extra features. It *should* build with 5.0.2 too, so you can try that ignoring the warning.

The error looks like a place where the script is not safe against spaces path names.

lakfel, 2022-04-14:
I did ignore and continue with the installation but now, the rendering does not work
[image]

I tried to run the test and a bunch of cgal test failed
[image]

t-paul, 2022-04-14:
That looks more like a GPU / OpenGL issue. What's in `Help` -> `Library Info`?

lakfel, 2022-04-14:
Here,

`Compiler: GCC "9.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.0.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.12.8
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20210627
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.6.4
cairo version: 1.16.0
lib3mf version: (not enabled)
Features: fast-csg, fast-csg-trust-corefinement, fast-csg-debug, fast-csg-exact, fast-csg-exact-callbacks, fast-csg-remesh, fast-csg-remesh-predictibly, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, sort-stl
Application Path: /usr/local/bin
Documents Path: /home/lakfel/.local/share
User Documents Path: /home/lakfel
Resource Path: /usr/local/share/openscad
User Library Path: /home/lakfel/.local/share/OpenSCAD/libraries
User Config Path: /home/lakfel/.config/OpenSCAD
Backup Path: /home/lakfel/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/home/lakfel/.local/share/OpenSCAD/libraries
/usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/lakfel/.local/share/fonts
/home/lakfel/.fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/home/lakfel/Library/Fonts
/usr/share/fonts/X11
/usr/share/fonts/cMap
/usr/share/fonts/cmap
/usr/share/fonts/opentype
/usr/share/fonts/truetype
/usr/share/fonts/type1
/usr/share/fonts/X11/Type1
/usr/share/fonts/X11/encodings
/usr/share/fonts/X11/misc
/usr/share/fonts/X11/util
/usr/share/fonts/cmap/adobe-cns1
/usr/share/fonts/cmap/adobe-gb1
/usr/share/fonts/cmap/adobe-japan1
/usr/share/fonts/cmap/adobe-japan2
/usr/share/fonts/cmap/adobe-korea1
/usr/share/fonts/opentype/urw-base35
/usr/share/fonts/truetype/dejavu
/usr/share/fonts/truetype/droid
/usr/share/fonts/truetype/noto
/usr/share/fonts/truetype/quicksand
/usr/share/fonts/truetype/ubuntu
/usr/share/fonts/type1/gsfonts
/usr/share/fonts/type1/urw-base35
/usr/share/fonts/X11/encodings/large
GLEW version: 2.1.0
OpenGL Version: 3.1 Mesa 21.2.6
GL Renderer: D3D12 (Intel(R) UHD Graphics 630)
[... 198 lines cut ...]
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays`

lakfel, 2022-04-15:
Yes it is WSL.

I could fix it. WSL was not using my gCard, installing CUDA from NVIDIA solved it. Thanks!

closed 2022-04-15 by lakfel