openscad/openscad #4230: Wrong focus handling for InputDevice events

t-paul, 2022-04-19:
With focus lost some events are blocked (rotate?) and some are still reported and handled (translate?).

Possible cause reported:
> https://github.com/openscad/openscad/blob/master/src/gui/input/InputDriverManager.cc#L163
> `event->activeOnly` might part is the cause, for the rotation events, it's always false

crazyimp, 2022-04-19:
With this patch, input events will not be handled when the window is no longer focused. Tested with spacenav input.
Not sure why there is an exception for "out of focus events".
```diff
diff --git a/src/gui/input/InputDriverManager.cc b/src/gui/input/InputDriverManager.cc
index 1dea006a3..be8b43ab0 100644
--- a/src/gui/input/InputDriverManager.cc
+++ b/src/gui/input/InputDriverManager.cc
@@ -160,7 +160,7 @@ void InputDriverManager::sendEvent(InputEvent *event)
 
 void InputDriverManager::postEvent(InputEvent *event)
 {
-  QWidget *window = event->activeOnly ? QApplication::activeWindow() : currentWindow;
+  QWidget *window = QApplication::activeWindow();
   if (window) {
     QCoreApplication::postEvent(window, event);
   }
```

Lenbok, 2022-05-24:
How will this effect dbus input though? See #3246 and https://github.com/Lenbok/scad-dbus which depends on being able send dbus input events to openscad while another window (external editor) has focus.

MichaelPFrey, 2022-05-24:
@Lenbok do you mean #3367?

The merge request you mentioned is customizer related, which should not interact with InputDevice and/or DBUs.

Lenbok, 2022-05-24:
@MichaelPFrey thanks, I had a typo, I meant to reference #3426 (have updated the comment), but yes #3367 is the corresponding issue.