openscad/openscad #4236: Compilation error

mbracconi, 2022-04-23:
I am compiling openscad (specifically 2020.12-branch) on a CentOS cluster with the NULLGL=ON with gcc-9.2.0

I get these errors:
`In file included from /software/energia2/tools/openscad/openscad/src/CGALRenderer.h:3,
                 from /software/energia2/tools/openscad/openscad/src/export_png.cc:10:
/software/energia2/tools/openscad/openscad/src/renderer.h:21:46: error: ‘shaderinfo_t’ in ‘class GLView’ does not name a type
   21 |  virtual void draw_with_shader(const GLView::shaderinfo_t *) const  { this->draw(true, true); }
      |                                              ^~~~~~~~~~~~
/software/energia2/tools/openscad/openscad/src/renderer.h:51:60: error: ‘shaderinfo_t’ in ‘class GLView’ does not name a type
   51 |  virtual void setColor(const float color[4], const GLView::shaderinfo_t *shaderinfo = nullptr) const;
      |                                                            ^~~~~~~~~~~~
/software/energia2/tools/openscad/openscad/src/renderer.h:52:59: error: ‘shaderinfo_t’ in ‘class GLView’ does not name a type
   52 |  virtual void setColor(ColorMode colormode, const GLView::shaderinfo_t *shaderinfo = nullptr) const;
      |                                                           ^~~~~~~~~~~~
/software/energia2/tools/openscad/openscad/src/renderer.h:53:84: error: ‘shaderinfo_t’ in ‘class GLView’ does not name a type
   53 |  virtual Color4f setColor(ColorMode colormode, const float color[4], const GLView::shaderinfo_t *shaderinfo = nullptr) const;
      |                                                                                    ^~~~~~~~~~~~
In file included from /software/energia2/tools/openscad/openscad/src/CGALRenderer.h:3,
                 from /software/energia2/tools/openscad/openscad/src/export_png.cc:10:
/software/energia2/tools/openscad/openscad/src/renderer.h:57:124: error: ‘shaderinfo_t’ in ‘class GLView’ does not name a type
   57 |  virtual void render_surface(shared_ptr<const class Geometry> geom, csgmode_e csgmode, const Transform3d &m, const GLView::shaderinfo_t *shaderinfo = nullptr) const;
      |                                                                                                                            ^~~~~~~~~~~~
In file included from /software/energia2/tools/openscad/openscad/src/export_png.cc:52:
/software/energia2/tools/openscad/openscad/src/ThrownTogetherRenderer.h:15:38: error: ‘shaderinfo_t’ in ‘class GLView’ does not name a type
   15 |  void draw_with_shader(const GLView::shaderinfo_t *shaderinfo) const override {
      |                                      ^~~~~~~~~~~~
/software/energia2/tools/openscad/openscad/src/ThrownTogetherRenderer.h:18:38: error: ‘shaderinfo_t’ in ‘class GLView’ does not name a type
   18 |  void draw_with_shader(const GLView::shaderinfo_t *, bool showedges) const;
      |                                      ^~~~~~~~~~~~
/software/energia2/tools/openscad/openscad/src/ThrownTogetherRenderer.h:22:68: error: ‘shaderinfo_t’ in ‘class GLView’ does not name a type
   22 |  void renderCSGProducts(const CSGProducts &products, const GLView::shaderinfo_t *, bool highlight_mode, bool background_mode, bool showedges,
      |                                                                    ^~~~~~~~~~~~
In file included from /software/energia2/tools/openscad/openscad/src/export_png.cc:52:
/software/energia2/tools/openscad/openscad/src/ThrownTogetherRenderer.h:24:75: error: ‘shaderinfo_t’ in ‘class GLView’ does not name a type
   24 |  void renderChainObject(const class CSGChainObject &csgobj, const GLView::shaderinfo_t *, bool highlight_mode,
      |                                                                           ^~~~~~~~~~~~
make[2]: *** [CMakeFiles/OpenSCAD.dir/src/export_png.cc.o] Error 1
make[1]: *** [CMakeFiles/OpenSCAD.dir/all] Error 2`

Could you help me in fixing this issue?

Thanks

t-paul, 2022-04-23:
Cherry picking the fix from `master` (https://github.com/openscad/openscad/commit/864443dd7bdd7bc958fbce6cadd1e313f8d0a52f) does not work unfortunately. So unfortunately it's unlikely to be fixed in the old branch without a clear motivation of this being critically needed.

mbracconi, 2022-04-24:
I managed to compile the `master` branch with some modifications to enable custom location of the external libraries.
Thanks

t-paul, 2022-05-01:
What's that custom code, would that make sense upstream? Do you know that there's `OPENSCADPATH` as environment variables?

mbracconi, 2022-05-02:
I have basically modified the CMakeLists.txt to provide the location of the `double-conversion` library and to add the path of the includes of the `gmp` and `mpfr` libraries along with some other minor modifications in the `cmake/Modules` folder
That has been necessary since I haven't root permission on the machine and I have manually installed all the compulsory libraryies outside the standard location. 
Could it be interesting for you?

t-paul, 2022-05-02:
It's not currently required for the builds I maintain, but I'd be happy to include such changes that also may help others with building. Maybe we can go through in small separate changes, starting with the separate stuff from `cmake/Modules`.

t-paul, 2022-05-15:
I'm closing the issue here, as said above, feel free to create PRs for the cmake changes. If they are not making things much more complicated, I'm happy to merge those.

closed 2022-05-15 by t-paul