openscad/openscad #4239: I'd love to see a go-like include for external files from git

benzvan, 2022-04-26:
Rather than using submodules in my openscad repo, I would like to be able to use syntax like this to import from external repositories similar to the way golang allows imports from external sources.

```
include <git@github.com:openscad/MCAD>
```

include, use, could support this syntax. Adding support to import() would be a bonus.

t-paul, 2022-04-26:
First step would be package management similar to proposal in https://github.com/openscad/openscad/issues/3479 - how things move after that remains to be seen.

jordanbrown0, 2022-04-26:
I would think in terms of `include <url>` rather than some custom syntax.

benzvan, 2022-04-27:
I think using a URL is valid too. I used the git syntax as an example because I think a lot of people store their code in git and there are libraries to handle all the authentication with git tokens. Using a URL would mean you could only use publicly accessible repositories or would need to handle the user@password syntax securely.