openscad/openscad #4246: RFC - More tessellation control

zougloub, 2022-05-02:
`$fn`, `$fs`,  `$fa` are allowing to control the number of fragments in a shape; their goal is to control the deviation between the tessellated geometry and the ideal geometry, or to simplify rendering.

I would like to propose something which extends their functionality without changing the code too much, eg. using special values for them; it's a bit "dirty" but see, for example:

```C
if (fn == -1.0) {
  /*
    fs means max. error between polygon edge and ideal shape.
    eg. circle is tessellated using regular polygon with vertices on the circle;
    fs means max. error between polygon edge and circle;
    fa is unused.
  */
  return int(ceil(M_PI / ( acos((r + fn) / r) )));
}
else if (0) {
  /*
   Leaving room for future $fn perversion here.
  */
}
else {
  return Calc::get_fragments_from_r(r, fn, fs, fa);
}
```

t-paul, 2022-05-02:
Problems with that:
* Special case for *some* places, or even more tricky to apply to other things like fonts
* Introducing magic values is usually not good
* Interferes with automatic passing along of `$fX` values if other built-ins are supposed to keep there non-magic behavior

I'm leaving a strong vote against this kind of change. There should be better ways of giving more/finer control over tesselation.

zougloub, 2022-05-02:
I concur with your analysis @t-paul hence the RFC mention and title.

The quoted implementation of a new `get_fragments_from_r()` only handles `circle()` (and sphere/cylinder), and of course all shapes should be handled for each new convention.

I didn't formulate clearly enough that ensuing discussions would be about:
- The fact that there could be several definitions of “tessellation error”;
- Introduction of “max chord deviation considering segments on the original ideal shape” error control modality, which seems more relevant considering the goal of error control;
- Better ways than `fn/fa/fs` to control the error, vs. using magic negative values of `$fn` to encode different error modalities and up to 2 control values.

coryrc, 2025-08-11:
Why not proposing a new special `$fe` for "means max. error between polygon edge and ideal shape".

I'm going to write this PR soon, as this is exactly what I want. I want the error to be a fair bit smaller than the line width of my printer and I don't care at all what degrees or number that is.

coryrc, 2025-08-14:
> Why not proposing a new special $fe for "means max. error between polygon edge and ideal shape".

Heh, looking through the call stack, I see why now.

coryrc, 2025-08-15:
/assign-me

coryrc, 2025-09-30:
Could this be implemented in a way which works for third-party libraries which implement non-builtins? I.e. what if someone uses polyhedron, could we intercept accesses to $fn, $fa, and $fs and provide different values? ... I don't think so.

jordanbrown0, 2025-09-30:
> I.e. what if someone uses polyhedron, could we intercept accesses to $fn, $fa, and $fs and provide different values?

I'm not sure exactly what you're thinking of, but I have long thought that we should have a function that returns the number of segments for a circle of the specified radius/diameter, taking all of the `$f*` variables into account.

That is, `fn(r=10)` would return the number of segments that would be used for `circle(r=10)`.  What you then do with the information is up to you.

jordanbrown0, 2025-09-30:
In fact, I think I implemented that recently, though I don't remember exactly why.

coryrc, 2025-09-30:
> I'm not sure exactly what you're thinking of, but I have long thought that we should have a function that returns the number of segments for a circle of the specified radius/diameter, taking all of [...]

Hardest part of that is picking a name.

Also, welcome advice on naming this internalcode. Today we pass `fn, fs, fa` as individual arguments and store individually all over and call `get_fragments_from_r(r, fn, this->fs, or node.fa)` in many variations, which gets replaced by a single argument/member of this class + call a function with just `r` to get the number of segments/subdivisions/fragments/what-have-you.

```
class TessellationControl
{
public:
  TessellationControl(const Parameters& parameters, const ModuleInstantiation *inst=nullptr);
  std::optional<int> circular_segments(double r) const;
  friend std::ostream& operator<<(std::ostream& stream, const TessellationControl& f);
private:
  double fn, fs, fa;
};
```

jordanbrown0, 2025-09-30:
I would suggest `fn()`.

For the name of your class/functions... TessellationControl seems too broad; there are lots of aspects of tessellation that it doesn't control.  I would think of something that more directly brings in the words "circle", "arc", "segments", and/or "fragments".

nophead, 2025-09-30:
It is called r2sides(r) in NopSCADlib.

coryrc, 2025-09-30:
> Another alternative would to allow the user to implement `get_fragments_from_r`.

Only problem I've hit with this idea is the CSG intermediate format. It would be a big project to allow this.

coryrc, 2025-09-30:
Problem with refactoring so most of the code doesn't need to know what parameters control discretization:

Python functions like `python_circle` are doing:

```
  char *kwlist[] = {"r", "d", "fn", "fa", "fs", NULL};

  if (!PyArg_ParseTupleAndKeywords(args, kwargs, "|ddddd", kwlist, &r, &d, &fn, &fa, &fs)) {
```

to get out fn, fa, and fs as args, so every function has to know about every possible option because the above is written to allow in Python: `circle(1.2, 2.4, 1, 2, 3)` to set all of r, d, fn, fa, and fs. Which, personally, seems bad: specifying both r and d is not good, but, relevantly, fn/fa/fs should have been keyword-only. I can make that latter change to make them keyword-only and then only `ArcDiscretizer` needs to know about the params. How does this sound?

 [Docs for PyArg_ParseTupleAndKeywords](https://docs.python.org/3/c-api/arg.html#c.PyArg_ParseTupleAndKeywords),

coryrc, 2025-09-30:
@gusohler do you have an opinion on making fn/fa/fs (and possible future discretization controls) keyword-only args?

(details in previous post)

jordanbrown0, 2025-09-30:
Yes, I agree that `fn()` is awfully general.  But it's a **really** good match for the existing practice.

For general use, I'd say that "discretization" is too non-common-English.  But for internal use, shrug.

$fn/$fs/$fa **are** used to control spheres, so they are at least somewhat used in 3D contexts.  And I expect that if we ever did Bézier patches those variables would have a role, though it wouldn't be as easily defined as the role for circular stuff.

For CSG output, there needs to be an interface back out that turns your object back into a list of input variables - perhaps as a string holding a comma-separated list, or perhaps even via an `operator<<` to a stream.

jordanbrown0, 2025-09-30:
To be clear:  I have no opinion on how this fits into the Python API; I haven't looked at it.

jordanbrown0, 2025-10-01:
I was curious about the number of faces on the spheres.  What's directly calculated is the number of vertices.  It's easiest for me to think of it as lines of latitude and longitude.  There are $fn lines of longitude, so the equator has the same number of segments as a $fn circle.  There are $fn/2, rounded up, lines of latitude, because you only walk across half of the circle.  The number of vertices is the product of those two, so (for instance) $fn=7 yields 7 lines of longitude times 4 lines of latitude is 28 vertices.

But there are only faces **between** the lines of latitude, so that's one fewer ring, so it's 7 times 3 is 21, plus the two polar faces is 23.

coryrc, 2025-10-01:
> You can't _quite_ only write out $fn, because it looks like there are a couple of cases (notably in extrusion) where it does some calculation based on the size of the child geometry, which isn't knowable at CSG time.

Ah, darnit, I hadn't gotten that far yet, am still at the trivial cases. You're talking about `extrudePolygon()` in `linear_extrude.cc`.

> But CSG is just a subset of OpenSCAD-language, so the fact that "everything" needs to know how to process them doesn't really mean much - they already need to know. Theoretically we might reduce to $fn for the cases where we can, and then a CSG consumer might not need to know as much as a full OpenSCAD-language processor does, but it looks like there are still a few cases where it does.

It's my understanding it's used as the "lossless" export to FreeCAD and maybe others in the future, so if I add the original request of this Issue -- i.e. set segment count so error is just below some target -- they may need to be updated.

Except... dur, the point is FreeCAD *doesn't* need to care about these parameters because it can represent the true shape? (except when circle($fn=) is used for polygons?) So it should be fine for it to ignore any new segmentation specials?

So for the purposes of this Issue I think it's as simple as writing out `$fe` everywhere `$fs` is written for CSG?

MichaelAtOz, 2025-10-01:
The [wiki section ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Circle_resolution:_$fa,_$fs,_and_$fn)is peppered with the word 'fragments', so regardless of LLM OpenSCAD speak should perhaps be **fragments(r)** (?) But...
The wiki also has:
 "Or you can embed this OpenSCAD version in your code to work out what's going on, you need to set r= to your size
   echo(n=($fn>0?($fn>=3?$fn:3):ceil(max(min(360/$fa,r*2*PI/$fs),5))),a_based=360/$fa,s_based=r*2*PI/$fs);"

Which IIRC I added, does it really need a built-in?

coryrc, 2025-10-01:
> does it really need a built-in?

I proposing to change it, so everybody with a copy will need to update, so why not have it be available as a builtin? They're trivially shadowed, so it's not like it'll cause problems.

jordanbrown0, 2025-10-01:
CSG rules are murky.  The only thing that's truly promised is that a current OpenSCAD can read its own CSG output, and probably that it can read all previous OpenSCADs' output.  Emitting $fe in place of previous values - if you can even figure out what the equivalent is - would definitely break reading the CSG file on all existing consumers, or at least all existing ones that do circles right.

The rationale for adding a built-in is so that we can tweak the algorithm and have programs able to cope without having to be upgraded.  (Past needing to use the new function, of course.)  Why would we want to tweak the algorithm?  To add new ways to control the accuracy, as discussed here.  To improve it, e.g. to force $fa/$fs to generate multiples of four, so that the circle always touches all four sides of the bounding box.

MichaelAtOz, 2025-10-01:
Damn markup that is r * 2 * PI

`echo(n=($fn>0?($fn>=3?$fn:3):ceil(max(min(360/$fa,r*2*PI/$fs),5))),a_based=floor(360/$fa),s_based=floor(r*2*PI/$fs));`

jordanbrown0, 2025-10-01:
Sorry, I read a previous comment as writing $fe instead of $fs, when you meant that anywhere we write $fs we should **also** write $fe.

I also realized that we **must not** simply calculate and emit $fn, as appealing as that might be.  If you do that, the consumer can't tell whether or not the original had $fn set.  A reasonable interpretation would be that an explicit $fn requests a regular polygon while no explicit $fn requests a circle, and always emitting $fn would break that interpretation.

coryrc, 2025-10-03:
> The [wiki section ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Circle_resolution:_$fa,_$fs,_and_$fn)is peppered with the word 'fragments',

I've proposed a change to the page to fix that.

coryrc, 2025-10-22:
I have a number of AIs in [#6251](https://github.com/openscad/openscad/pull/6251) to work on, have been working on color instead. Still will be working on this.

coryrc, 2025-11-07:
Color spec is done, coming back to this soon.

jordanbrown0, 2025-11-24:
I would be a little concerned that somebody might expect to set `$fe` to control curve resolution, and then be surprised when it *doesn't* control some of the cases.  I don't think it needs to do exactly the right thing; I think it just needs to do something that's sort of reasonable.

Ref #6059, my opinion (which is not universally shared) is that the intent of $fa/$fs/$fe is to produce an approximation of a circle, and the details of that approximation are not all that important.  ($fn is a different animal; at least for the circular stuff it needs to produce exactly what's specified.)

coryrc, 2026-08-24:
> (IIRC, this performance hit is not for the circular cases, but rather is for linear extrusion with twist?)

Only extrusion with twist *and* scaling. Simple solid geometry are constant-time calculations (sphere, cylinder).

> Would anybody ever notice the difference between a truly correct (and computationally expensive) calculation, and one that is not quite as "correct" but is near-zero cost?

The latter is impossible for the edge cases. You're effectively just asking for the default settings, which is just to divide the rows and columns into a grid, and has no guarantees at all about error. Anything better is computationally demanding to varying degree.