openscad/openscad #4249: nothing provides 'libboost_filesystem.so.1.78.0()(64bit)'

mdarge, 2022-05-03:
On **openSuse** openscad is included to the main repository. But since some days openscad won't start anymore. During new installation it is says:
`Problem: nothing provides 'libboost_filesystem.so.1.78.0()(64bit)' needed by the to be installed openscad-2021.01-4.2.x86_64
 Solution 1: do not install openscad-2021.01-4.2.x86_64
 Solution 2: break openscad-2021.01-4.2.x86_64 by ignoring some of its dependencies`

Why does openscad insists on version 1.78.0()(64bit) ?
As the libboost package is under active development things will only getting worse. Only 1.79 and the older 1.76 are available now.

t-paul, 2022-05-03:
For openSUSE main repository, please open tickets there. This is not maintained here.

closed 2022-05-03 by t-paul

mdarge, 2022-05-03:
It seems to be a problem of openscad and is independent of the Linux distribution. Distributions only pull available code. If openscad is working with any libboost greater than some version the problem is solved. The problem is openscad demands for 1.78 and nothing else.

t-paul, 2022-05-03:
This is a dependency issue. "Problem: nothing provides" means OpenSCAD was built against something not available via your package manager and that is specific to the distribution/installation. That has nothing to do with the source code, and there's is nothing that can be done here.

OpenSCAD does not require boost 1.78, this specific build was created with that dependency. For example the nightly build happily works on Ubuntu 18.04 building with boost 1.65.1.

Why the link to Ubuntu? The first message says openSUSE?

mdarge, 2022-05-03:
Ok, it's the difference between the source and the binary. I don't have the courage to compile for my own. And Ubuntu only because they also have no 1.78 library. You're right. Sometimes openSUSE gives the choice to downgrade a package, not this time.

t-paul, 2022-05-03:
Maybe it gets solved on the openSUSE side. In the meantime, you can try using the AppImage from the [download page](https://openscad.org/downloads.html) which is meant to be a bit more independent of distributions, although that comes at the price of other potential problems.