openscad/openscad #4251: Feature Request: Warning for semicolon after translate

jbinvnt, 2022-05-03:
In the following example:
```
translate([1,2,3]);
cylinder(h=1,r=2);
```
at first it might look like the cylinder is going to be translated. But no translation would occur because there is a semicolon after `translate(...)`. The current behavior in OpenSCAD is to accept this code silently. I suggest that a warning is added any time a semicolon directly follows `translate` so that the user is not caught off-guard by this behavior.

jordanbrown0, 2022-05-04:
Probably the form that this would take would be for some built-in modules to warn if they are called with no children.  (It would be kind of the inverse of the built-in modules like cube() that warn if they are called *with* children.)

jordanbrown0, 2022-08-16:
```
translate(p);
```
is equivalent to
```
translate(p) { }
```
in that they both have no children.  (Just as `cube(){}` is not an error.)

Would it be a problem if translate (and some similar modules) warned on `translate(p) {}`?

The only non-error case I can think of is if somebody had all of the child objects commented or *'d out.  Useless, but not meaningless.

MichaelAtOz, 2022-08-16:
It is legal syntax. 
Getting into "your program construct is stupid" is going to far IMO.
Things like that, or other strange usage, are commonly employed during debugging.

If it were to progress it would need to only log once, as such could be in loops or modules called many times.

Maybe flagging such in the editor with say a colour highlight?

jordanbrown0, 2025-03-08:
There may also be an issue with an invocation where the syntactic children do not generate any geometric children, e.g.

```
translate([1,2,3]) { if (false) cube(); }
```
... which should *not* generate a warning, but might depending on the implementation technique used.