openscad/openscad #4254: Editor: comment unused lines

Gregor-Gregor, 2022-05-06:
Add a menu command to comment when executed all the lines containing unused code (eg. not used variable definitions, unreferenced modules etc.).

Gregor-Gregor, 2022-05-06:
Purpose: to help cleaning up the complex code from all the dead ends, temporary solutions etc.

MichaelPFrey, 2022-05-07:
I need a minimum example, so...
```
if(0){
    echo("you should not see this");
}else{
    echo("run always");
}
```
More interesting is
```
a = true;

if(a){
    test1();
}else{
    test2();
}

//never run because overwritten
module test1(){
    cylinder();
}

module test1(){
    cube();
}

module test2(){
    sphere();
}
```
Detecting unused lines is a hard problem.
   * `a` is picked up by the customizer.
   * `a` can be overwritten, when the file is included.
 
 Much easier, but similar would be just to indicate lines that have actually been interpreted/run/visited.

I have a rough idea for one possible implementation: We could add a flag "visited"/"run" to every ASTNode. Every ASTNode already has a Location. Just run the code as normal, just adding a flag when the Node was visited. We could then walk trough the AST-Tree and add a green background to every line (or even character - we have the column) that ran. By not resetting the color until the code is changed, multiple runs (triggered by customizer or animation) could accumulate.

I have following main concerns:
   * much runtime impact would such a feature have?
      * adding and setting a bool should be cheap
      * having the feature "default off" should avoid the biggest isssue
      * by doing the UI changes after render/preview, it should feel like an unnecessary slow down
   * is there any way to setup an automated test?
      * this describes a GUI feature, but requires core work
      * would a "run lines of code" count help?
   * how robust is our location source code?
      * hidden bugs in location could be uncovered during the work - which is a good thing - 

I take a look into it

t-paul, 2022-05-07:
I'm not sure the "comment lines" is a good strategy. That could make things even harder if there's for example multiple assignments on one line.

It might be better to go with the logic other editors use by highlighting the issue in some way (marker on the sidebar, underlining the assignment, something like that).

I think using the AST is probably the way to go, meaning it needs at least the "compile" run. Traversing that should not be a huge concern, but having a feature toggle is probably still a good idea.

At this point there's no GUI tests, it might be possible for this scenario that we extend the AST export to include the extra information so it can be referenced via expected files. There's actually a number of features that could do with a better AST handling, like some simple optimizations and collecting timing information.

I very much agree with the potential location issues, it should have improved quite a bit, but there might be still some problems lurking and finding those along the way would be good.

Gregor-Gregor, 2022-05-07:
_Hi Michael,_ 

I put a very rough and simplistic description of an idea I had, and you dug into it amazingly. :)

First of all, my description to "comment unused lines" was in fact inaccurate expression to describe the first concept, meaning, in fact, to "mark unused code" of the code somehow (I'll edit the title to reflect this more precisely). Your idea to use a text marking (background or text color, any other formatting available) is much better, especially that it doesn't require tricky changes to the original code. Marking of "unused" code may be just a simple negation of marking the "used/interpreted/run/visited" code.

Now, another great idea is to allow the result to cumulate over multiple runs. The same code should usually generate the same effect of such a marking, unless there is one of the cases you mentioned, but also possibly any use of _rands()_ in decision making.

I absolutely agree the future should be normally disabled, working only on specific request of the user.

MichaelPFrey, 2022-05-09:
here a prove of concept:

https://github.com/MichaelPFrey/openscad/tree/mfr-2022-05-07-forrest

[image]

The Indicator code works. The color needs to be changed to be controlled by the color profile. The isEvaluated Logic seam to work. I need to rewrite the AST Tree navigation.