openscad/openscad #4255: Feature request: is_empty(geometry) function to test for empty geometry

Gregor-Gregor, 2022-05-07:
The idea is to have a built-in function returning boolean _true_ when the _geometry_ given as a parameter is degenerated to nothingness.
It would be great for 2D geometry, but I believe it's quite easy to implement also for a 3D one.

Quite a logical extension would be e.g. overlap testing function:

`function do_overlap(geometry1, geometry2) = is_empty(intersection() { geometry1; geometry2; });
`

I'm aware we don't mix geometry with mathematics just yet, so the syntax above would be kinda novelty (esp. giving the geometry as a parameter to the function).
But it would be extremely useful - imagine parametrizable filling of the 2D shape with custom pattern, when you'd like to elliminate all degenerated elements at the edges. I can also imagine an establishing of geometry size or bounding box with help of this function. 
I believe the similar result can be achieved even now by some complex math analysis of the geometry _we_ create, but when it comes to importing 2D or 3D files - there's no helper available I know about.

t-paul, 2022-05-07:
This seems to be way too far off from what's possible with the current engine. I think we need to first get the stepping stones sorted (e.g.  #3956) so I'm inclined closing this for the time being.

Gregor-Gregor, 2022-05-07:
Ah, https://github.com/openscad/openscad/pull/3956 contains a solution to the problem defined here, indeed.

I read there:
> data contains these fields:
>     min : [x,y,z] containing the minimum of the bounding box.
>     max : [x,y,z] containing the maximum of the bounding box.
>     center : [x,y,z] containing the center of the bounding box.
>     size: [x,y,z] containing the size of the bounding box.
>     **points: [[x1,y1,z1],[x2,y2,z2]...[xn,yn,zn]] containing the points/vertices of the polyhedron or polygon**.
>     faces: in case of 3d geometry [[a1,a2,...,an], [b1,b2,...,bn],....[z1,z2,...,zn]] containing the point indices of the faces (can be trianges, quads or larger polygons).
>     paths: in case of 2d geometry [[a1,a2,...,an], [b1,b2,...,bn],....[z1,z2,...,zn]] containing the point indices for the outlines.

Having this information allows to check for empty geometry like this:

` (len(data.points)==0)`

(whatever the syntax to reach the _points_ field is).

Thanx!

Gregor-Gregor, 2022-05-09:
It crossed my mind (and I bet somebody requested it already), the nice functionality would be to have a geometry as a data (object) type, so we can calculate it _one time only_, then reference multiple times.

for example, currently:

```
difference() {
  scale([2,2,2]) complex_geometry();
  complex_geometry();
}
```

generates 2 instances of the same complex_geometry(),
while

```
cg = complex_geometry();

difference() {
  scale([2,2,2]) cg;
  cg;
}
```

would calculate it only once.
Also accessing some properties of such a geometrical object would be possible,
like _cg.points_ from an example above.

MichaelAtOz, 2022-05-13:
Geometry is cached, so in the example, `complex_geometry()` only gets evaluated once, not multiple times. So from that perspective this is not needed.

However `cg = complex_geometry();` so you can process `cp` is being looked at.

nophead, 2022-05-13:
Yes geometry is cached but expression values are not, they get computed each time and are used as the key to the cache. So, for example, if you compute a polyhedron of a spiral spring using sweep the polyhedron is cached but to re-uses it you have to run the sweep again to calculate all the points or store it in a variable. Then you run into the issue of variables being recalculated in used modules.

t-paul, 2022-05-15:
Objects / Geometry as data is discussed in #3087.

closed 2022-05-15 by t-paul