openscad/openscad #4257: Need editor tool to comment/uncomment selection

williamlugg, 2022-05-09:
It would be extremely useful if the OpenSCAD editor had a tool that would allow the user to comment or uncomment a block of selected code.

My thought is that the tool would work as follows:

1. User selects lines of code. (We'll assume they're uncommented for the sake of this discussion)
2. User selects comment/uncomment block tool.
3. The tool adds the comment characters to the beginning of all the selected lines.

Alternatively:

1. User selects block of code previously commented using tool.
2. User selects comment/uncomment tool.
3. The tool removes the comment characters from the commented lines.

This would be very useful for troubleshooting a model with a problem - parts of it could be easily disabled temporarily for testing purposes. 

Thanks for listening. 
Bill lugg

thehans, 2022-05-09:
There is already Comment and Uncomment under the "Edit" menu, with hotkeys Ctrl-D and Ctrl-Shift-D, respectively.

closed 2022-05-09 by thehans

t-paul, 2022-05-09:
Also check https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Modifier_Characters#Disable_Modifier which is a structural comment instead of the clasic text based `//` and `/* */`.

MichaelAtOz, 2022-05-13:
For posterity, having icons for this would make it much more usable. For us non keyboard worriers ([sic](https://en.wiktionary.org/wiki/sic))