openscad/openscad #4259: Does transparency work incorrectly or is it intended that way?

kberkut, 2022-05-13:
Does transparency work incorrectly or is it intended that way?

[image]
[image]

LookoutHill, 2022-05-14:
It appears that I am seeing the same issue - transparency is only one way (or so it would appear from my model). One picture is the view from above (the subfloor is opaque) and the other is seen from below (the subfloor is transparent). The responsible snippet is: color(alpha = 0.1) { /* Omitted the code to place plywood panels, but it contains no other color manipulations or % operators. */ }
[image]
[image]
.

jordanbrown0, 2022-05-14:
@kberkut @LookoutHill cut your problem case down to a minimum demonstration and supply the program, please.  Transparency is quite limited and perhaps you are hitting its limits.

Here's a program that demonstrates an ... interesting ... behavior that is more wrong than I expected:

```
translate([0,0,0]) cube(10);
translate([0,0,10.1]) color(alpha=0.2) cube(10);
translate([0,0,20.2]) cube(10);
```

Looking down, it's kind of what you'd expect:
[image]

Looking up, however...
[image]
Note that you can see through the upper cube.

I'm not so much surprised that there are artifacts, as that they are asymmetric.

jordanbrown0, 2022-05-14:
A bit of memory refresh and experimentation later...

Order matters.  Have the transparent objects be after the non-transparent objects.

```
translate([0,0,0]) cube(10);
translate([0,0,20.2]) cube(10);
translate([11,11,10.1]) color("blue") cube(10);
translate([-11,-11,10.1]) color("blue") cube(10);
translate([0,0,10.1]) color(alpha=0.3) cube(10);
```
works pretty well.  Moving the transparent cube to the top:
```
translate([0,0,10.1]) color(alpha=0.3) cube(10);
translate([0,0,0]) cube(10);
translate([0,0,20.2]) cube(10);
translate([11,11,10.1]) color("blue") cube(10);
translate([-11,-11,10.1]) color("blue") cube(10);
```
... not so well.

This is the source of the asymmetry I observed in my previous experiment.

jordanbrown0, 2022-05-14:
I believe that the reason for this is that preview basically operates by drawing all of the faces and letting the display driver and hardware figure out which ones are closest to the camera.  A transparent face lets some of the existing image through.

So if you draw the solid object first, and then the transparent object that is in front of it, the transparent faces let the solid object show through.

If, on the other hand, you draw the transparent object first, there is nothing to show through, and then when you draw the solid object behind it, well, it's behind something existing and so doesn't show.

None of this is "right", of course, but it's how the high-speed preview stuff works.

kberkut, 2022-05-14:
> @kberkut @LookoutHill Same result.

`translate([0,0,0]) color("red") cube(10);
translate([0,0,10.1]) color(alpha=0.1) cube(10);
translate([0,0,20.2]) color("green") cube(10);`

[image]
[image]
[image]
Linux ARCH 5.17.5-arch1-2

Another simple example.
`$fn=100;
module any_tube(diametr_outer, height, thinkness){
  rotate_extrude(convexity = 10) 
  translate([diametr_outer/2-thinkness,0,0])
  square([thinkness,height]);
}
module move(x=0,y=0,z=0,rx=0,ry=0,rz=0)
{translate([x,y,z]) rotate([rx,ry,rz]) children();}
color("red", 0.2) any_tube(160,300,0.5);
color("blue") cylinder(300,55,55);
`
[image]
[image]

kberkut, 2022-05-14:
[image] .
Cool) I designed a frame house in sketchup. If you are interested, please contact)

jordanbrown0, 2022-05-14:
Swap the cylinder and the any_tube to put the transparent object (the any_tube) last, and you get this:

[image]

Similarly for the green/transparent/red stack, if you put the transparent cube after the green cube, you get this:
[image]
Note that we are seeing the bottom of the green cube, rather than seeing through it.

Looks to me like this is a duplicate of #1390, but I don't have the authority to close it.

kberkut, 2022-05-14:
Confirm. It is enough to swap places. But still, this is wrong behavior. This is a bug.

[image]

jordanbrown0, 2022-05-14:
Yep.  Ref #1390.  Known for a long time.  Very hard to fix.

kberkut, 2022-05-14:
Alternatively, add an entry about this feature of OpenSCAD to the help section.

"For the correct display of transparent objects, the location of instructions (modules) should be from the inside of the model to the outside."

But unfortunately this is not always possible, especially in complex models.

kberkut, 2022-05-14:
[image]

$fn=100;

module any_tube(diametr_outer, height, thinkness){
  rotate_extrude(convexity = 10) 
  translate([diametr_outer/2-thinkness,0,0])
  square([thinkness,height]);
}

module move(x=0,y=0,z=0,rx=0,ry=0,rz=0)
{translate([x,y,z]) rotate([rx,ry,rz]) children();}

// The correct location.
color("red", 0.2) any_tube(10,300,0.5);
color("blue", 0.2) any_tube(30,300,0.5);
color("green", 0.2) any_tube(50,300,0.5);
color("pink", 0.2) any_tube(70,300,0.5);

// The incorrect location.
move(x=100) color("red", 0.2) any_tube(10,300,0.5);
move(x=100) color("pink", 0.2) any_tube(70,300,0.5);
move(x=100)color("green", 0.2) any_tube(50,300,0.5);
move(x=100)color("blue", 0.2) any_tube(30,300,0.5);

jordanbrown0, 2022-05-14:
See the note I just added to #1390 about a problem that can't be solved with ordering.

LookoutHill, 2022-05-14:
@jordanbrown0 @kberkut Sweet! Moved the subfloor code to the end of the program and all is well. Thanks! A note in the documentation near % and color() would go a long ways towards managing this issue.

jordanbrown0, 2022-05-14:
I added a caution to the [documentation for color()](https://en.wikibooks.org/w/index.php?title=OpenSCAD_User_Manual/Transformations#color):

Warning: alpha processing (transparency) is order-sensitive. Transparent objects must be listed after non-transparent objects to display them correctly. Some combinations involving multiple transparent objects cannot be handled correctly. See issue [#1390](https://github.com/openscad/openscad/issues/1390).

(I don't understand the publication process for the documentation; it might not be immediately visible to others.)

kberkut, 2022-05-14:
> I added a caution to the [documentation for color()](https://en.wikibooks.org/w/index.php?title=OpenSCAD_User_Manual/Transformations#color):
>
> Warning: alpha processing (transparency) is [...]

I think the example with the mutual arrangement of pipes will be more understandable and friendly for new users. Including the picture.

t-paul, 2022-05-14:
Closing as duplicate of #1390.

closed 2022-05-14 by t-paul

MichaelAtOz, 2022-05-15:
Note also #310.