openscad/openscad #4263: color entanglement on non-intersecting object

skeptonomicon, 2022-05-22:
Two non intersecting objects of different colors will blur the colors across objects.
In one example, two panels form a tenon joint. Both panel's tenons become a mixture of blue and green. They should be rendered the same color as the panel they are part of. 
[miter_joint_scad.txt](https://github.com/openscad/openscad/files/8750152/miter_joint_scad.txt)

[image]

UBaer21, 2022-05-22:
This is [Z-fighting](https://en.wikipedia.org/wiki/Z-fighting) or 
[explained in FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?)

closed 2022-05-22 by t-paul

UBaer21, 2022-05-22:
it is possible to have clear colors if the objects are not generated by overlapping cubes with cutouts  but via a polygon
[image]

UBaer21, 2022-05-22:
another fix is to use color("blue") render() object  - that way the real geometry is generated so no intersection results in preview
[image]

skeptonomicon, 2022-05-27:
First, thanks for the quick response and resolution. I am still trying to understand this from an end user perspective. Is it the green subtracted tenon that colors the blue surface or is it the green original cube (before the subtraction) that colors the blue surface. From an end user perspective, there is a expectation that a union of two objects should indeed be the sum of the two objects and not a merge of the sub-objects. If the render somehow fixes this problem with the difference, should not the difference code include that portion of render that brought about the fix.

nophead, 2022-05-27:
No because render() calls CGAL to calculate the result of the union or difference, which is very slow. Preview just draws the faces in the correct order to simulate the union or difference. It doesn't calculate the resulting geometry.

UBaer21, 2022-05-27:
the original big cubes overlap or more exactly their faces share the same location. If you move them just 0.01 a color can be assigned only to that face without the other.
[image]
you still see z-fighting on that small cube at the top because i didn't moved them in Z only moved skyBlue in Y and the other in X.  You can also use something like  translate( $preview?[-0.01,0,0]:[0,0,0])  to ensure more exact values in render. 
This happens because preview is not a real 3D view it is a 2D view that looks 3D - a trick / illusion  - that way drawing is much faster. Your Objects have just a few vertices (cube = 8) so using render() makes only little difference.  If you have a complex scene a render could take minutes or even hours, you don't want this in preview for every little change.

skeptonomicon, 2022-05-27:
Ah, so the whole effect is a compromise for speed in the preview.