openscad/openscad #4264: Can't open file in the WSL file system

mkaisercross, 2022-05-23:
If you try to open file such as \\wsl$\Ubuntu\home\myuser\test.scad it just hangs.

t-paul, 2022-05-23:
Please provide details as schown in [FAQ#How_do_I_report_bugs?](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?) and following sections.

mkaisercross, 2022-05-23:
The OpenSCAD version: 2021.01
The Operating System name and version: Windows 10 Home, 21H2
A description of the scenario that produces the issue:
1. Install Ubuntu on WSL - https://docs.microsoft.com/en-us/windows/wsl/install
2. Start OpenSCAD
3. Click "open" button
4. Choose a file in the WSL filesystem. e.g. \\wsl$\home\myuser\example.scad

Expected: Model opens as normal
Actual: The file dialog closes and you go back to the splash page

In case of graphics issues, the OpenGL driver information: N/A
If possible, a trimmed down script reproducing the issue: N/A