openscad/openscad #4265: Sigma Integrated Rule Set fail?  SecureAge APEX: Malicious??

Logic-Elliven, 2022-05-23:
Hi.
Just being (over) cautious here by running the installer :
https://files.openscad.org/OpenSCAD-2021.01-x86-64-Installer.exe
through the signed up/in version of VirusTotal:
https://www.virustotal.com/gui/file/fbe1e590e1af2af863f5d92bd263eb4644d0ee29352e3ab566366e87a8f6ed12/details

"1 match for rule Execution File Type Other Than .exe by Max Altgelt from Sigma Integrated Rule Set (GitHub)
Checks whether the image specified in a process creation event **doesn't refer to an .exe file** (caused by process ghosting or other unorthodox methods to start a process)"

"1 match for rule Execution Of Not Existing File by Max Altgelt from Sigma Integrated Rule Set (GitHub)
Checks whether the image specified in a process creation event **is not a full, absolute path** (caused by process ghosting or other unorthodox methods to start a process)"

Then SecureAge APEX finds the file to be "Malicious"

I am aware of how difficult it is the get software to 'pass' every AV, trying to stand out as 'better' and am way out of my depth when it comes to the Sigma Integrated Rule Set
So:   ???!
:)

t-paul, 2022-05-23:
So, unless there's more information, I don't see any possible action here. I'm not fighting against an "AI powered" scanner that just says "something is wrong" when all other scanners see no issue.
Public VirusTotal check does not seem to include that check and is all green - https://www.virustotal.com/gui/url/aa682fda8635933ff343d64653148da19a52ff7798bcad27a541909d5ba1467a/detection

The source code of OpenSCAD is public, so if anyone has additional ideas, we can have a look.

closed 2022-05-23 by t-paul

Scopeuk, 2022-05-23:
looking at the messages above my guess would be that it has taken exception to the .com file we use to launch the exe in terminal mode (first message) and is taking exception to the method used to invoke the exe from the com (second message). On that basis I would assert this is likely a false positive

t-paul, 2022-05-23:
Good point, so that would be about the command line wrapper handling the stdin/stdout stuff:
https://github.com/openscad/openscad/blob/6aae7963498675dfb5bedaaef2cb8e56bdcddcab/winconsole/winconsole.c