openscad/openscad #4266: text, btt/ttb settings make certain fonts not horizontal align correctly

jglauche, 2022-05-23:
I don't know the bug is with the fonts or in openscad; however if you try to align a text going in Y direction (btt or ttb) you may get a different horizontal alignment depending on the font. 

`text(text="Meow",valign="top",halign="left",direction="btt");`
This seems to be a problem with all Liberation fonts on my system. halign = left pretty much centers the font. 
[image]

Another test case for me is:
`text(text="Meow",valign="top",halign="center",direction="btt", font="Liberation Sans");`
This puts the font at the Y center (towards -Y direction)
[image]

Not all fonts have issues with this, for example Helvetica works as I expect:
`text(text="Meow",valign="top",halign="left",direction="btt", font="Helvetica");`
[image]

t-paul, 2022-05-23:
Oh, I have not seen this type of extreme result, but it seems to be fixed with the alignment changes that came with #3684. With the nightly build it looks ok on my system (Debian/Testing).
The overall alignment looks quite different now, I guess that needs another review if there's not more/other issues.

Note that the alignment is still based on font information, not geometry. With the new (experimental feature) `textmetrics()` ([docu](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/WIP/Text_Metrics)) it's possible to query that font information giving more options for alignment.

jordanbrown0, 2025-03-08:
I think this was addressed as part of the textmetrics() work.

closed 2025-03-08 by jordanbrown0