openscad/openscad #4268: DXF export in the latest development snapshot produces possibly incorrect files

JanKalin, 2022-05-26:
Using the latest development snapshot, 2022.05.15.ci11888 (git 6aae79634) on Windows 10, the DXF files are apparently incorrect. Steps to reproduce:

- Create a simple script that produces a square: `square(10);`
- Render with *F6* and export with *File|Export|DXF*
- Load into Autodesk's DXF viewer https://viewer.autodesk.com/designviews

The viewer complains that the file is invalid: 
[image]

Repeating the same steps with the stable version 2021.01 works OK. Possibly the problem is in the Autodesk's viewer, but a colleague of mine also tried importing it into CorelDraw and it failed. I've attached the two files produced by different versions.

[Square_DXF.zip](https://github.com/openscad/openscad/files/8782496/Square_DXF.zip)

t-paul, 2022-05-26:
It would help if there's also a recovered version of the file, assuming that works as mentioned in the screenshot.

JanKalin, 2022-05-26:
I absolutely agree. But I don't have access to AutoCAD or I would have tried it. Perhaps someone with access can try - the file is `Square_developer_snapshot.dxf` in the attached ZIP.

t-paul, 2022-05-27:
Yes, the change for that is #3869.

OratHelm, 2024-06-08:
I think I have the same problem. With the latest snapshot development 2024.06.07 (macOS 14.5, Apple M2) exporting to DXF produces files that cannot be opened in AutoCAD 2024. AutoCAD opens in a black window with no cursor and closes when any command is used. Without any message.

The stable version (2021.01) produces files that open correctly, but they are made up of sets of lines instead of polylines, which is a problem for me.

The files in the JanKalin ZIP archive behave in the same way. I once got the following message when the window closed after trying `Square_developer_snapshot.dxf` (but I haven't managed to get this error message again...):
[image]
I tried to delve into these files to find out what the problem was, but I couldn't find anything and my knowledge of the subject is very limited.

jordanbrown0, 2024-06-10:
While it sounds like OpenSCAD is generating a bad file, it is also wrong that AutoCAD misbehaves that badly when given a bad file - a report to them is in order. It might be a security bug.

t-paul, 2024-07-30:
@apr-nbx Thanks, that's very useful. Would you be able to generate a trivial DXF example file with just a simple LWPOLYLINE plus maybe one or 2 other objects and attach this here together with a PDF print how it should look like? With that input maybe someone could try fixing the issue without having access to Autodesk themselves.

OratHelm, 2024-07-30:
@apr-nbx 
The DXF files generated by development snapshot (I tried with 2024.06.07) do contain a header, unlike those in the stable version which open correctly in AutoCAD.

In the header, the format version is specified, but the one specified seems to me to be too old to support LWPOLYLINE (`AC1006`, i.e. R10). However, simply replacing it with a more recent version (`AC1032` for AutoCAD 2018, `AC1027` for AutoCAD 2013 or `AC1014` for R14) is not enough to open the file in AutoCAD 2024.1.3. So there may be a problem in the header, but it's not that simple.

Here's what it looks like:
```
  0
SECTION
  2
HEADER
  9
$ACADVER
  1
AC1006
  9
$INSBASE
 10
0.0
 20
0.0
 30
0.0
  9
$EXTMIN
 10
0
 20
0
  9
$EXTMAX
 10
1
 20
1
  9
$LINMIN
 10
0
 20
0
  9
$LINMAX
 10
1
 20
1
  0
ENDSEC
  0
```

richfelker, 2024-09-09:
I just encountered this today trying to submit DXF exports to SendCutSend for cutting. Oddly, I still have the DXF exports for an earlier version of the project I submitted last year, and their web interface *did* correctly interpret them then, but rejects them now.

peterzieba, 2025-04-21:
Using the stable version, I've encountered issues with Adobe Illustrator being able to open a .DXF generated  by OpenSCAD, which I've narrowed down to the header being missing.

I've prepended this to the generated .DXF and it seemed to fix things:
```
  0
SECTION
  2
HEADER
  9
$ACADVER
  1
AC1009
  0
ENDSEC
```

It sounds like a lack of header is creating problems in other situations as well?

kintel, 2025-04-21:
We do write that header, but we use `AC1006` as the value for `$ACADVER`. Are you suggesting that the root cause is that this version is too old and that tools like Adobe Illustrator only supports newer files (e.g. `AC1009`) ?

peterzieba, 2025-04-21:
No, I'm saying stable, "OpenSCAD version 2021.01" does not create any manner of header. As in:

$ cat test.dxf | grep -i header
$ cat test.dxf | grep -i ver

both yield nothing.

kintel, 2025-04-21:
Oh, 2021.01 is ancient - please try a recent development snapshot. A lot of work has been done since then.

kintel, 2025-04-21:
This particular issue was resolved in #3869

peterzieba, 2025-04-21:
Ok, on "OpenSCAD version 2025.04.21.nightly" I receive the following entirely new error:

[image]

This time I couldn't figure out a solution as to how to modify the file.

What I will say, and this could be a red herring, is I use de-caff viewer sometimes just to view files, and it has a "messages" tab where it tells me what is on its mind. In this case:
"Entity type LWPOLYLINE is not defined for this DXF file's version (first occurance in line 108)!"

Again, that might not have anything to do with why Illustrator doesn't like it, but it did help me identify the "HEADER" issue from OpenSCAD stable.

peterzieba, 2025-04-21:
So, it looks like the resulting file in Nightly has arcs as opposed to bunches of tiny connecting lines on things that have been rounded / fillet radius. This is incredible and solves lots of problems for me.

Unfortunately in my current case, I'm dealing with this laser cutter. I'd rather not touch Illustrator if I can avoid it, but it is this particular laser cutter which insists on this as a workflow and so with Nightly the file is completely unopenable unless I can figure out what is going on.

peterzieba, 2025-04-21:
This seem to be related to https://github.com/openscad/openscad/issues/3989

"The frogurt is also cursed." "That's bad." "But you get your choice of toppings."

That issue shouldn't have been closed. There really is something to this that is breaking the ability to open the generated .DXF in nightly that impacts multiple pieces of software.

kintel, 2025-04-21:
This isn’t an easy problem to solve, unless someone can come up with a way of determining whether a DXF file is valid. In particular if the issue is isolated to commercial sw which cannot easily be tested.

zaitcev, 2025-04-22:
There's an easy solution: back out all the broken changes done since 2021.01. Roll all of it back. Problem solved.

kintel, 2025-10-07:
@dklaskowski Are you able to reproduce the issue using an extremely simple design (e.g. a square)?

dklaskowski, 2025-10-07:
projection() 

  difference() {
  
    translate([-60,150/4,0]) cube([120,90/4,5]); // filler between base and ports 

// add bolt holes for mounting the top and bottom together

    for(i=[0:1:7]) {
      translate([i*15-210/4,390/8,-1]) cylinder(7,d=3); // yes it is 3mm because it is laser cut! 
    } // end for

  } // end difference

dklaskowski, 2025-10-08:
I thought that a rectangle with six holes was pretty simple, but just for a test, I tried this:

projection() cube();

and the resulting file failed to import into Sketchup with the same error about some geometry being extremely far from the model design...

dklaskowski, 2025-10-08:
Again -- do the same in the 2021.01 version of OpenScad and it all works fine.

dklaskowski, 2025-10-08:
I exported the projected cube from the 2021.01 release verions of OpenScad and from the October 5th nightly build into the following files. I hope someone can sort it out and fix this! NOTE: I added the .txt file extension so they could be included in this post... Rename them without the .txt extension and you will be back to the original exported file. (They are ASCII files at heart).

[Square from nightly 1005.dxf.txt](https://github.com/user-attachments/files/22757315/Square.from.nightly.1005.dxf.txt)
[Square from release 2021.01.dxf.txt](https://github.com/user-attachments/files/22757316/Square.from.release.2021.01.dxf.txt)

kintel, 2025-10-08:
I suspect this to be a bug in SketchUp's DXF importer.

That said, I believe the issue is that SketchUp doesn't support the `AcDbPolyline` subclass marker (see spec https://help.autodesk.com/view/OARX/2024/ENU/?guid=GUID-748FC305-F3F2-4F74-825A-61F04D757A50).

This subclass marker was introduced in #3869, so before removing it, we need to check that we don't introduce any regressions against other common software ingesting DXF.

t-paul, 2025-10-08:
I don't think it should be just removed without other option. It was specifically asked for as some workflows break without it. As that PR shows, it was tested with 5 different open source applications. If it really is a bug is Sketchup, it should be fixed there.

That said, I'm not 100% sure it's a Sketchup bug. It could be that the OpenSCAD export does not flag the file version correctly. As it also, from the initial post, fails in the web viewer, it's probably more an issue with the generated file.

Worst case, we could add an export option, or see if there's some export library that could be used. (Didn't someone claim a desire to add some library everywhere some years ago? I guess that motivation died down or did not reach here).

OratHelm, 2025-10-08:
> I suspect this to be a bug in SketchUp's DXF importer.

Given the number of softwares that cannot open DXF files generated by the latest versions of OpenSCAD, I do not think the problem lies with the Sketchup import function but rather with OpenSCAD.
As I mentioned above, AutoCAD cannot open these files (error ‘Exception in AcDbSmartCenter.dbx ARX Command’, more details above), and neither can Autodesk's DXF viewer (mentioned here by @JanKalin), Adobe Illustrator (mentioned here by @peterzieba), NanoCAD 5 (#3989) or SendCutSend (mentioned here by @zaitcev).

Also, if it can help, I shared in #3989 a 100x100 square made in AutoCAD 2024 and exported in DXF, as a comparison with the same shape made in OpenSCAD.

kintel, 2025-10-08:
Does anyone know of any (free/OSS) SW that can read and write DXF that can exhibits this issue?

dklaskowski, 2025-10-08:
This won't help, but the free DXF Viewer in the Microsoft App Store can read and display the nightly files just fine.

AND LibreOffice Draw opens the good .dxf file just fine while rendering the broken .dxf file as text.

dklaskowski, 2025-10-30:
I saved the file to my computer, renamed it without the .txt at the end and tried to open it in LibreOffice Draw. It thought it was a text file... If the file is a good, LibreOffice Draw will display the object stored in the .dxf file.

So, my rough conclusion is that the file is broken.

dklaskowski, 2025-10-30:
Okay -- I moved the file to my graphic system and tried to import it into Sketchup and got the same errors as before: Some imported geometry is extremely far from the model origin.

So, per my testing and playing (see my comments above), the file you sent is no good AND Sketchup and LibreOffice Draw both agree that it is no good.

I also tried to load it into Xometry for a quote. It did not like the file and gave this error: File square.dxf contains geometry detected as not flat in the X-Y plane.

The file is broken.

jordanbrown0, 2025-10-30:
I don't have any diagnostic tools for DXF files, but they're text so I took a look, aided by the documentation at
https://help.autodesk.com/view/OARX/2024/ENU/?guid=GUID-235B22E0-A567-4CF6-92D3-38A2306D73F3

One thing struck me as potentially relevant: The $INSBASE is specified as an XYZ tuple.  It's 0,0,0, but it's still a 3D coordinate in an otherwise 2D file.

t-paul, 2025-10-30:
There's more wrong, for starters https://ezdxf.readthedocs.io/en/stable/dxfentities/lwpolyline.html indicates that the version tag is wrong. But that alone also does not fix things.

kintel, 2025-10-30:
Try removing the AcDbEntity entries; I suspect that to be unsupported in the apps in question ..

dklaskowski, 2025-10-30:
Note for all that are trying to fix this: If you rename the file with .txt and edit it, you can delete any or all lines you think are making the problem. Then rename the file back to a copy without the .txt and try to open it in LibreOffice Draw to see if you get a square up on the screen... If it instead opens LibreOffice Writer and displays text, it is probably still broken...

t-paul, 2025-10-30:
And again as I said before, do NOT use LibreOffice as reference as all the files above work here with LibreOffice just fine. So that test is not relevant at all as there seem to be other things at play.

jordanbrown0, 2026-02-25:
Perhaps inspecting the Inkscape implementation, or asking the Inkscape folks for help, might be useful.