openscad/openscad #4269: Customizer slider control is too small in Windows 10.

ScottNudds, 2022-05-27:
The OpenScad customizer slider control used in both the current official release and the current build is unusable because the slider handle is only 2 or 3 pixels tall, and only extends above the central line.

In addition no tick-marks appear, no upper and lower values appear, 

This problem makes the customizer practically worthless.

t-paul, 2022-05-27:
Please add at least some more details like screen resolution (e.g. is it a 4k screen? are there any special display settings), and a screenshot..

ScottNudds, 2022-05-28:
Window Resolution 2048x1152
Latest build of Openscad and the last release build.
Windows 10 Home 64 bit.
scale and layout 100% (default)
[image]

ScottNudds, 2022-05-28:
Consider this issue resolved.  Turns out it was one a Theme that was installed that was causing the issue.

Reverting back to the default theme restored the slider handles to their appropriate mode and size.

closed 2022-05-28 by thehans