openscad/openscad #4270: Lookup unnecessarily limited

ScottNudds, 2022-05-28:
Lookup <key,value> array

Value is limited to numbers.  Doesn't support vectors.

Lookup <key,Vector> array

t-paul, 2022-05-28:
Lookup [interpolates between values](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#lookup), how would that work with vectors?

jordanbrown0, 2022-05-29:
I'm not sure it's what the OP is thinking of, but I wanted this (and implemented my own) some time ago.

```
vals = [
    [ 1, [10,10,10] ],
    [ 2, [20,20,20] ],
    ...
];
v3 = lookup(1.5, vals);
```
would yield `[15, 15, 15]`.

Exactly as `((1.5-1)/(2-1))*([20,20,20] - [10,10,10]) + [10,10,10]` does.  Straightforward interpolation, but where the values are vectors.

The input vectors would have to be add/subtract compatible with one another, but other operations (notably add and subtract!) already have that restriction.

I used it to interpolate between key positions and rotations in an animation.

UBaer21, 2022-05-29:
maybe you are looking for [Search](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Search)

UBaer21, 2022-05-29:
@jordanbrown0  I have a "map(from=[0,1],to=[45,90])" function that works with vectors for "to" (only without constraining) I think this would be more universal to implement.

MichaelPFrey, 2022-05-29:
assuming @ScottNudds wants what @jordanbrown0  describes, "lookup unnecessarily limited" is surprisingly accurate (the algorithm remains the same and the hard part for implementing this feature - multiplying a vector with a double - is already implemented).
I have [prove of concept code](https://github.com/MichaelPFrey/openscad/tree/2022-05-29-lookup) and the main works looks to be writing tests, cleaning up the code and testing. Some edge cases could also be tricky.

I also see the use case (animation interpolation) as described by @jordanbrown0.

Sample Code for the feature:
```
n = 7;

function pointOnCircle(r,a,x,y) = [r * cos(a) + x, r * sin(a) + y];

a = [ for(i=[0:1:n]) [i/n, pointOnCircle(500,i*360/n*5,0,0)]];

echo(a);

 function get_cylinder_h(p) = lookup(p, a);

 color("black")
 for (i = [0:0.005:n]) {
    // echo(i, get_cylinder_h(i));
    echo(i);
    translate(get_cylinder_h(i)) cylinder(r1 = 6, r2 = 2, h = 30);
 }
```

Sample Screenshot of the prove of concept:

[image]

ScottNudds, 2022-06-17:
Michael Frey seems to have clued in.  Now will his code be accepted?
Probably not.

t-paul, 2022-06-17:
The example looks reasonable and useful. I think we probably need to have warning messages for cases that are not supported and a clear definition of the extended support.

PinkPandaKatie, 2024-02-09:
I ran into this issue recently and implemented a workaround by converting the original table from `[[in_value, out_value], ...]` to `[[in_value, index], ...`, calling `floor(lookup(...)` on that, then using that index to look up the two adjacent values from the original table, then lerping between those. It seems to work as expected but I haven't tested every edge case.

```
function _lerpv2(sv, ev, st, et, t) = t <= st ? sv : t >= et ? ev : sv + (ev - sv) * ((t - st) / (et - st));
function _lerpv1(t, v1, v2) = _lerpv2(v1[1], v2[1], v1[0], v2[0], t);
function _lerp_lookup(t, values, idx) = _lerpv1(t, values[idx], values[min(idx + 1, len(values) - 1)]);

// Fixed version of lookup() which works with vectors
function vector_lookup(t, values) = _lerp_lookup(
    t, values,
    floor(lookup(t, [for (i = [0 : len(values)]) [values[i][0], i]]))
);
```

amatulic, 2024-11-15:
I was just about to create a new issue report for exactly this problem. I was trying to use `lookup()` to find interpolated values in a table of distances (keys) and 2D coordinates (values). The return is always `undef`.

Given that one can do simple math operations already with vectors, I see no reason why an interpolation between vectors wouldn't be possible in `lookup()`.

amatulic, 2024-11-15:
For those who run across this issue, the following is a substitute function for `lookup()` that works with vectors. I made this as efficient as I could, but I suspect it would be more efficient if it was all done internally like `lookup()`.
````
/*
Vector lookup function
by Alex Matulich
November 2024

Same functionality as lookup() but works with vector values instead of numeric values, because the built-in lookup() doesn't work with vectors.
The key is searched in the 0 column of the lookup table.

Parameters:
needle = value to look for
haystack = lookup table, in the form [[key, vector], [key, vector] ...]
Keys must be sorted low to high.
The vectors can be any length but must all be the same length.

Returns a vector interpolated from the lookup table.
A search exceeding the maximum key in the table causes the last vector to be returned.
A search for lower than the minimum key in the table causes an error.
*/

// demo
echo(vectorlookup(11, [
    [0,[1,1]],
    [4,[3,2]],
    [8,[5,3]],
    [12,[7,4]],
    [16,[9,5]]
]));

// return interpolated vector from searching a table of lookup values and vectors.
function vectorlookup(needle, haystack) = let(
    n = len(haystack),
    ilow = binarysearch(needle, haystack, n, 0, n-1),
    ihigh = ilow < n-1 ? ilow+1 : ilow,
    frac = (needle-haystack[ilow][0]) / (haystack[ihigh][0]-haystack[ilow][0]),
    ) haystack[ilow][1] + frac * (haystack[ihigh][1]-haystack[ilow][1]);

// binary search of 2D table
function binarysearch(needle, haystack, n, low, high) =
    high < low ? -1 :
    let(mid = low + floor(0.5*(high-low)))
        mid >= n-1 ? n-1 :
        (haystack[mid][0] <= needle && needle < haystack[min(mid+1,n-1)][0] ? mid :
        (haystack[mid][0] > needle
            ? binarysearch(needle, haystack, n, low, mid-1)
            : binarysearch(needle, haystack, n, mid+1, high))
        );
````

UBaer21, 2024-11-15:
is this what you try to archive?

```
p=[
[0,0],
[10,20],
[15,25],
[18,30],
[27,30],
[34,20]
];

x=[for(i=[0:len(p)-1])[i,p[i].x]];
y=[for(i=[0:len(p)-1])[i,p[i].y]];

for(i=[0:.1:len(p)-1])translate([lookup(i,x),lookup(i,y)])cube(1,true);
```

jordanbrown0, 2026-05-17:
I want to do this.