openscad/openscad #4271: Bounds missing on sliders

ScottNudds, 2022-05-28:
There is sufficient room above the slider bar to show the lower:step:upper bounds.
Not displayed.

Also tick marks are not displayed on the slider.  They are part of any standard slider control.

t-paul, 2022-05-28:
Please create issues that are self contained and fully explain what they are about. I assume from the earlier ticket that your refer to the Customizer as I've seen that on. That's not how it works with many people involved in open source projects, some maybe just picking a random issue to work on.
Just consider that any extra second you spend extra now on posting a nicely explained feature request increases the chance of it getting it implemented at some point.

ScottNudds, 2022-06-17:
You do know what bounds are, don't you?
You do know what a slider control looks like don't you?
Upper and lower bounds are missing from the slider control.

t-paul, 2022-06-17:
Closing as requested.

closed 2022-06-17 by t-paul