openscad/openscad #4272: Installing openscad on Amazon Linux 2 / Red Hat

reikje, 2022-05-29:
I am trying to create a Lambda layer to run openscad from an AWS Lambda function. I am struggling to install openscad using yum on an EC2 instance running Amazon Linux 2. Have tried to follow the steps from the Dockerfile here: https://github.com/openscad/openscad/issues/3738 but some things have changed since this was published. Ended up getting a compile error that was somehow related to boost.

Unfortunately, I am not a C expert to understand the root cause. Does anyone know where I could get a binary for either Amazon Linux 2 or Red Hat to try? Maybe there is a yum repository that I can add?

Thanks in advance,
Reik

thehans, 2022-05-29:
I don't know what repositories Amazon Linux 2 (or anything else about the OS), but OpenSCAD is available in official Fedora repositories, which uses yum, for example.
Have you looked over the myriad Linux install options on the downloads page?  http://openscad.org/downloads.html

Its not clear if you are looking for the most recent development snapshots (aka "nightlies"), or stable release only.
If dev snapshots are fine, then there are a number of repos for the various versions of fedora on the very bottom of that same download page:  http://openscad.org/downloads.html#snapshots
Maybe one of those is compatible?

Otherwise there are also AppImage, flatpack, and snap based packages which may work.

If none of those will do, you can still try building from source.  The dockerfile you linked to is using qmake build system, and we have since moved our builds entirely to cmake since then (again, depends if you are building from master HEAD or a specific release tag).
To use cmake you can replace the qmake commands with the compilation instructions in the README:
https://github.com/openscad/openscad#compilation
If you want further help with the building, you'll need to paste the actual logs showing the specific build errors you're getting.

Scopeuk, 2022-05-29:
The details requested above would allow a much better diagnosis but my first suspicion is this might be a repeat of https://github.com/openscad/openscad/issues/4202

reikje, 2022-05-30:
Thanks for the quick reply. At this stage, I am simply trying to compile any version. The steps I am running are all based on the `Dockerfile` in #3738 

@thehans when you write that `qmake` is no longer used, does that mean can get rid of the following lines:
```
RUN git clone https://code.qt.io/qt/qt5.git && cd qt5 && git checkout 5.12 && perl init-repository
RUN mkdir qt5-build && cd qt5-build && ../qt5/configure -confirm-license -developer-build -opensource -nomake examples -nomake tests
RUN cd qt5-build && make -j4 && make install
RUN export PATH=/root/qt5-build/qtbase/bin:$PATH
RUN echo 'export PATH=/root/qt5-build/qtbase/bin:$PATH' > ~/.bashrc
```

I have modified two things in the Dockerfile in #3738 
- `xy-utils` package is just called `xz`
- boost has moved to jfrog, so I changed to the following (also bumped to latest version): `RUN wget https://boostorg.jfrog.io/artifactory/main/release/1.79.0/source/boost_1_79_0.tar.gz && tar -zxf boost_1_79_0.tar.gz && cd boost_1_79_0 && ./bootstrap.sh && ./b2 install`  

The error I am hitting is below:

```
tar -xf CGAL-4.9.1.tar.xz && cd CGAL-4.9.1 && cmake . && make -j4 && make install
-- The CXX compiler identification is GNU 7.3.1
-- The C compiler identification is GNU 7.3.1
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
== Setting paths ==
-- Build CGAL from release in directory CGAL-4.9.1
-- Packagenames: CGAL-4.9.1
== Setting paths (DONE) ==

== Generate version files ==
-- CGAL_MAJOR_VERSION=4
-- CGAL_MINOR_VERSION=9
-- CGAL_BUGFIX_VERSION=1
-- CGAL_SONAME_VERSION=12
-- CGAL_SOVERSION     =12.0.0
-- CGAL_REFERENCE_CACHE_DIR=
-- Building shared libraries
-- Targetting Unix Makefiles
-- Using /bin/c++ compiler.
-- USING CMake version: 2.8.12
-- System: Linux
-- USING GCC_VERSION = '7.3.1 20180712 (Red Hat 7.3.1-14)'
-- Using gcc version 4 or later. Adding -frounding-math
== Generate version files (DONE) ==

== Set up flags ==
-- Build type: Release
-- USING CXXFLAGS = ' -frounding-math -O3 -DNDEBUG'
-- USING LDFLAGS = '  '
== Set up flags (DONE) ==

== Detect external libraries ==
-- External libraries supported: GMP;GMPXX;MPFR;ZLIB;OpenGL;LEDA;MPFI;RS;RS3;OpenNL;Eigen3;BLAS;LAPACK;QGLViewer;ESBTL;Coin3D;NTL;IPE
-- Preconfiguring library: GMP ...
-- Found GMP: /usr/lib64/libgmp.so  
-- GMP has been preconfigured:
--   UseGMP-file:      
--   GMP include:      /usr/include
--   GMP libraries:    /usr/lib64/libgmp.so
--   GMP definitions:  
-- USING GMP_VERSION = '6.0.0'
-- Preconfiguring library: MPFR ...
-- Found MPFR: /usr/lib64/libmpfr.so  
-- MPFR has been preconfigured:
--   UseMPFR-file:      
--   MPFR include:      /usr/include
--   MPFR libraries:    /usr/lib64/libmpfr.so
--   MPFR definitions:  
-- USING MPFR_VERSION = '3.1.1'
CMake Error at /usr/local/lib/cmake/Boost-1.79.0/BoostConfig.cmake:240 (cmake_policy):
  Policy "CMP0057" is not known to this version of CMake.
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindBoost.cmake:177 (find_package)
  cmake/modules/CGAL_SetupBoost.cmake:30 (find_package)
  cmake/modules/CGAL_SetupDependencies.cmake:85 (include)
  CMakeLists.txt:673 (include)

CMake Error at /usr/local/lib/cmake/Boost-1.79.0/BoostConfig.cmake:244 (if):
  if given arguments:

    "ALL" "IN_LIST" "Boost_FIND_COMPONENTS"

  Unknown arguments specified
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindBoost.cmake:177 (find_package)
  cmake/modules/CGAL_SetupBoost.cmake:30 (find_package)
  cmake/modules/CGAL_SetupDependencies.cmake:85 (include)
  CMakeLists.txt:673 (include)

-- Configuring incomplete, errors occurred!
```

thehans, 2022-05-30:
> Thanks for the quick reply. At this stage, I am simply trying to compile any version. The steps I am running are all based on the `Dockerfile` in #3738
>
> @thehans when you write that `qmake` is no [...]

No, I meant the lines at the bottom where openscad itself is built (and the line in the Dockerfile explicitly calls `qmake`).
```
RUN cd openscad && qmake openscad.pro
...
```
which should become something more like:
```
RUN cd openscad && mkdir build && cd build && cmake ..
```

The lines you pasted are for building Qt5, which is the GUI framework that openscad depends on.
*Although*, if you're using openscad only for its headless, command line interface(which I assume, from a cloud based service) then it should actually be possible to build without Qt5 via the `HEADLESS` build option.
eg: `cmake -DHEADLESS=ON ..`
and if you want the experimental features of the nightly then also include build option `-DEXPERIMENTAL=ON ` in the cmake call.

> I have modified two things in the Dockerfile in #3738
>
> * `xy-utils` package is just called `xz`
> * boost has moved to jfrog, so I changed to the following (also bumped to latest version): `RUN wget [...]

OK, so that is a build error during the build configuration step for the CGAL library, which is another *dependency of* OpenSCAD, NOT OpenSCAD itself.

The CGAL version from that Dockerfile (4.9.1) is out of date. The minimum required versions of all dependencies are listed in the [README here](https://github.com/openscad/openscad#prerequisites), which shows that we are on CGAL 5.4 now, so that will need to be updated.  I suspect the error is just some incompatibility between an older CGAL lib + a newer Boost library, and hopefully is resolved by simply updating.

Any further issues specifically with CGAL's build process would be better directed towards that project: https://github.com/CGAL/CGAL

closed 2024-02-04 by t-paul