openscad/openscad #4275: Broken preview

ygoe, 2022-05-30:
I have found a geometry that can't be rendered in the F5 preview. It works in the F6 rendering but that lacks any colour so it's really hard to recognise anything in there. This makes it very hard to work with this model at all!

The problems started when I added the `difference()` to the main profile part. That alone works well and I've used it before. But if I cut it at an angle of 45° like it's done in reality, this thing goes wild and produces random pixel garbage.

The shown structure is the environment I need to design a part into. That part goes into the profile so it's important to see its boundaries using colour. But as it is, it's impossible to see anything in the preview. And the full rendering doesn't have any colour so it's hard to see where my inner part ends and if it really fits into the profile.

OpenSCAD version: 2021.01
OS: Windows 10

Code:
```
color("#cccccc")
translate([-profileGap, 0, 2.35])
profile(60);

color("#ffcccc")
translate([0, profileGap, 2.35])
mirror([1, 0, 0])
rotate([0, 0, -90])
profile(60);

module profile(length) {
	difference() {
		// Main profile laid flat
		rotate([-90, 0, 90])
		linear_extrude(length)
		polygon([
			// 1. Steg: oben 4,2mm
			// Innen: alle 6,3mm, Wände: alle 1,0mm
			// Innen vom 1. zum 3. Steg: 21,2mm
			[1.5,2],[1.5,1],[0,1],[0,0],
			[4.2,0],[4.2,0.5],[3.7,1],[2.5,1],[2.5,2.5],
			// 2. Steg: oben 4,4mm
			[8.9,2.5],[8.9,1],[7.7,1],[7.2,0.5],[7.2,0],
			[11.6,0],[11.6,0.5],[11.1,1],[9.9,1],[9.9,2.5],
			// 3. Steg: oben 4,4mm
			[16.3,2.5],[16.3,1],[15.1,1],[14.6,0.5],[14.6,0],
			[19,0],[19,0.5],[18.5,1],[17.3,1],[17.3,2.5],
			// 4. Steg: oben 4,2mm
			[23.7,2.5],[23.7,1],[22.5,1],[22,0.5],[22,0],
			[26.2,0],[26.2,1],[24.7,1],[24.7,2],
			// Hauptprofil
			[26.2,2],[26.2,10],[24.2,10],[24.2,3.5],[2,3.5],[2,10],[0,10],[0,2]
		]);
		// 45° mitre cut
		translate([0, 0, -30])
		linear_extrude(40)
		polygon([
			[10,-10],[10,40],[-40,40]
		]);
	}
}
```

Preview:
[image]

Rendering:
[image]

t-paul, 2022-05-30:
See [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?), you need to set a `convexity` value for the `linear_extrude`. By the looks of it something between 3 and 5 should work for most model orientations.

closed 2022-05-30 by t-paul

MichaelPFrey, 2022-05-31:
you could also change
```
module profile(length) {
	difference() {
```
to
```
module profile(length) {
	render()
	difference() {
```

[image]

I would not recommend that for complex models, but for those simple profiles, this seams to do the trick.

jordanbrown0, 2022-05-31:
render() is documented:  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Render

Always look at the cheat sheet.

ygoe, 2022-05-31:
I know the cheat sheet and use it regularly from the help menu (I just didn't know the FAQ yet or consider this issue a FAQ). But for render() it says "Needs description."

jordanbrown0, 2022-05-31:
Ah, sorry, so it does.  I only read the first line, which says exactly what render() does.

The other application that it should mention is in avoiding preview artifacts, like when you haven't set a high enough convexity, or when you want to move the camera inside a large object used only for intersection or difference.

There should probably be an explicit discussion of the issues with preview, why they occur, and how to work around them.

jordanbrown0, 2023-10-28:
And there is such a discussion, at
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?

There are a number of troubleshooting tips in the FAQ.

(But it doesn't describe the "camera inside the model" problem, and should.)

jordanbrown0, 2023-10-28:
Added a section on camera-inside-the-model:  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_does_difference_(or_intersection)_sometimes_not_work_in_preview?