openscad/openscad #4276: Windows 11 ctrl shift d uncomment doesn't work

fredOnGitHub, 2022-05-31:
Hello,

I have seen ctrl shift d could do uncomment that but doesn't work. 

I don't see it in the edit menu.
[image]

How can add manually a shortcut to uncomment. 

Thanks

MichaelAtOz, 2022-05-31:
Please add [details listed here](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs?).
Ctrl-Shift-D is shown on my W7 & Debian systems.

jordanbrown0, 2022-05-31:
Note that the image shown drops out all of the Ctrl+Shift+Letter shortcuts, except for Ctrl+Shift+F.  It drops Z, I, C, and D.

jordanbrown0, 2025-03-08:
Appears to work on Windows 11 in 2025.02.28.

closed 2025-03-08 by jordanbrown0