openscad/openscad #4278: Remove material from imported STL breaks preview

hadess, 2022-06-06:
Using `openscad-2021.01-8.fc36.x86_64` and current Flathub version

In a number of my "editing" projects, I use OpenSCAD to chop off material using `difference()`, for example:
https://github.com/masible/extra-large-light-ruler-based-calipers

The preview appears with missing faces, but there are no errors in the logs. Removing the objects that remove material fixes the preview. The full on render doesn't show any artifacts

I understand that the STL files I've been using might have minor glitches, but the render can go past that and render correctly. I would happily take a slightly slower preview if that means I can preview with the colours I assigned rather than having to use the monochrome render.

Broken preview:
[image]

Working render of the same source:
[image]

Working preview, without the removed material:
[image]
.

nophead, 2022-06-06:
Have you see the convexity parameter of the import statement?

closed 2022-06-06 by thehans

hadess, 2022-06-06:
Setting the convexity to, say, 10, solves the problem. Is there any reason why a higher convexity than the default (I imagine it's 2) isn't the default? It only changes the preview, so should be a backwards-compatible change to make.