openscad/openscad #4280: High fragments causing hull in rotate_extrude

UBaer21, 2022-06-08:
```
union(){
rotate_extrude(angle=185,$fn=3881)translate([10,0])square();
cube();
}
```
`rotate_extrude(angle=185,$fs=0.009,$fa=0.09)translate([10,0])square();`
also causing this .. some angles doesn't 

version 2022.05.04 also  2022.06.01
[image]

kintel, 2024-04-01:
Status as of 2024-03-24:
* Still broken with CGAL
* Rendering works with Manifold, but F6-then-F5 results in an empty render

pca006132, 2024-09-05:
F6-then-F5 seems to be fine now with manifold. While I am testing with #5282, it should be unrelated?

kintel, 2024-09-19:
F6-then-F5 was fixed already before #5282. Lots of changes happened between march and now, so I won't try to understand why it's better :)

Status as of 2024-09-18 (pre-#5282):

Still broken with CGAL
Works with Manifold