openscad/openscad #4281: Feature request: modules should return objects

Anonymous-Coward, 2022-06-18:
I am drawing myself a window. I have started from the inside out. I first defined a window pane module with multiple transparent layers with air between them, spacers, and a frame - to which I pass parameters such as the overall height and width, the number of layers and the layer thickness (which are dictated by desired thermal performance and material characteristics) and the thickness of materials used. 

When I now want to draw the frame into which these panes should be placed, I have to re-compute manually the width of the pane, based on the number of layers and thicknesses of various materials and air gaps, which, inside the pane module, is computed by the module itself. That's not nice - whenever I need to change that calculation, I need to change it in two places.

Instead, what would be nice is that besides defining geometry, the call of a module should also have a return value which can be assigned to a variable, and the module should be allowed to define properties of the return value, rather than just local variables. Then, I could refer the pane's width from the geometry of the frame.

More generally, such a feature would be useful when designing various things composed of non-trivial parts - parts to which you specify some dimensions, but which compute other dimensions based on non-trivial calculations, where these computed dimensions are needed in other places of the design.

Of course, there's the obvious workaround of defining a function which should then be used by both the module and the place that needs the module's dimensions. It would still not be as nice and expressive as modules returning objects with computed properties.

In order not to have assignment of module calls to variables, which could make them difficult to distinguish from function calls, a syntax like:

         module(width, height, radius) : rectangleOne

could be used, where `module(width, height, radius)` defines some geometry, but also creates a variable called `rectangleOne` in the scope in which the module is called, to which the properties defined by the module are attached. As far as I can think of it, this could work on both object modules and operator modules.

t-paul, 2022-06-18:
See "object" discussion... #3088.

closed 2022-06-18 by t-paul

Anonymous-Coward, 2022-06-19:
@t-paul  So won't there be a release of OpenSCAD before OpenSCAD2 is released? I was thinking this is a fruit hanging low enough to be added before OpenSCAD2.

t-paul, 2022-06-19:
There's no such thing as a "OpenSCAD2" release. This is a proposal for new features and quite a number of them have been implemented. So there's no relation to OpenSCAD releases at all.

For example the data structure part of objects is already in the nightly builds, e.g. for `fontmetrics()` and JSON import. Reason for moving forward very slowly is just the limited time of people who are interested in working on this.

Anonymous-Coward, 2022-06-19:
I see. Thanks. I was under the impression that OpenSCAD2 is an initiative for a full rewrite, meant to happen in the not so near future. Sorry for the noise.

t-paul, 2022-06-19:
No worries, even just keeping the discussion alive helps sometimes. Showing more / different use cases can making things clearer.