openscad/openscad #4282: Slider control poorly implemented

ScottNudds, 2022-06-20:
There is sufficient room above the slider bar to show the lower:step:upper bounds.
They are currently not displayed.

Also tick marks are not displayed on the slider. They are part of any standard slider control

[image]

That is what a proper slider control looks like for the retarded.

MichaelAtOz, 2022-06-20:
@scottnudds Do you realise how offensive your posts seem?

UBaer21, 2022-06-20:
@ScottNudds you really need to adjust your attitude! 
If you search "GUI Slider" you will notice there are a lot of valid versions - so it is important you understand that your opinion of a "proper" slider is just subjective. 
In oSCAD slider are just a customizer GUI and represent always 0-100% of the defined range the designer sees appropriate - if you need a specific value you wouldn't choose a slider as a design element but a spin box (which is already there).

So why would you need the end start or end values?
Also in your example you can't see the actual value set - so why does the numbers matter then?

ScottNudds, 2022-06-20:
> @ScottNudds you really need to adjust your attitude! If you search "GUI Slider" you will notice there are a lot of valid versions - so it is important you understand that your opinion of a "proper" [...]

I see the problem.  You are stupid. Lazy, and don't know what a slider is.

ScottNudds, 2022-06-20:
> "Do you realise how offensive your posts seem?"

Do you realize how little respect I have for the retards who are pretending to be OpenSCAD developers?

t-paul, 2022-06-20:
Ok, that's enough.