openscad/openscad #4283: No efficient way to select from a number of multiple modules.

ScottNudds, 2022-06-20:
No rational; way to select from a large number of optional modules.

Problem:  There are 50 or more small modules that draw some simple shapes that are to be optionally embossed onto a surface.

OpenSCAD provides no facilities for cleanly selecting between the various modules without running through a series of if statements.

The solution would be to provide a function that executes a module based on an index and an array of module names.

Dispatch(N=module number, [list of module names]]

MichaelAtOz, 2022-06-20:
```
// call a specific module in a set of modules

dispatch(1);
_dispatch(2) { m1(); m2(); m3(); };

module m1() translate([0,0,0]) cube(1);
module m2() translate([-10,0,0]) sphere(1);
module m3() translate([10,0,0]) cylinder(3,2,1);

module dispatch(m=0)   {
  _dispatch(m=m) {
  m1();
  m2();
  m3();
  }
}

module _dispatch(m=0) {
  echo($children);
  children(m);
}
```

t-paul, 2022-06-20:
Long term solution for this is discussed in #3088 - rough goal is basically to make modules first class objects so they can be assigned to variables and passed around, similar to what's possible with functions now.

I'd be open to even have a separate built-in dispatch style feature, what other languages call "pattern matching", so if there's specific suggestions for that, we can open a separate discussion, otherwise please follow #3088 for updates.

Solution with current features is shown above, other slightly different example can be found in https://github.com/openscad/openscad/blob/master/resources/icons/svg-default/icons.scad

closed 2022-06-20 by t-paul

jordanbrown0, 2022-06-20:
Michael gives a clever solution to the problem the OP asked, but the OP didn't ask the right question.

Better than any index-based scheme is to pass the shape as a child.  The embosser shouldn't know the list of shapes that can be embossed; it should emboss whatever shape it's given.

```
// create a cube with the child shape embossed into it.
// child's origin is the center of the top of the cube.
// Portion of child below the Z=0 plane will be cut into the cube.
module emboss(dims) {
    difference() {
        cube(dims);
        translate([dims.x/2, dims.y/2, dims.z+0.001]) children();
    }
}

// Circle
emboss([10,10,10]) cylinder(h=1, d=3, center=true, $fn=30);
// Triangle
translate([20,0,0]) emboss([10,10,10]) cylinder(h=1, d=3, center=true, $fn=3);
```