openscad/openscad #4286: Improvement: Preview of surfaces with coplanar differences is inconsistent

DeflateAwning, 2022-06-23:
As can be seen in the following example, the coplanar surface where the difference is renders poorly in the preview. Would be great if this sort of thing rendered better.

Example Code:
```
difference() {
    cylinder(d=10, h=10);
    cylinder(d=5, h=10);
}
```

Preview Example:
[image]

Real Rendering Example:
[image]

OpenSCAD Version: 2021.01

Scopeuk, 2022-06-23:
This is unfortunately a mathematical ambiguity (a 3d shape with zero width in one dimension) left over by the Boolean operation
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?

UBaer21, 2022-06-23:
@DeflateAwning  the so called Z-fighting is a result from the faster preview display you can circumvent this by either make the differencing part bigger ```cylinder(h=100,d=5,center=true);```so it overlaps the surface. Or use ```render()``` in front of the difference

DeflateAwning, 2022-06-23:
I wonder if there's a potential solution where OpenSCAD could detect the coplanar surfaces, and then automatically apply a very, very small epsilon value to the height to avoid this issue automatically? 

I haven't dug too deep into the feasibility of this, but I'm curious if anyone has any thought on whether this is potentially do-able, or verging on basically impossible to fix.

UBaer21, 2022-06-23:
Design without overlap shouldn't be done as in more complex designs when rotations are involved you end up with float numbers that are not the exact mathematical solution. That will lead to render errors. Coplanar faces are detected not sure if it would be wise to move them automatically to repair bad designs.

DeflateAwning, 2022-06-23:
I can't think of a negative outcome of moving coplanar faces to repair nonoptimal designs. Can you?

UBaer21, 2022-06-23:
There are the preview problem (z-fighting) which has to do how far your camera position is and FoV ($vpf) angle. 
The other is when rendering. 
Z-fighting is also a good way to see if objects share  same faces when align them so i see this as a feature. 
For rendering it may reduce some errors but you get a different geometry and as units are not defined it is impossible to determine the amount for a deviation.

DeflateAwning, 2022-06-23:
I've been using OpenSCAD for a very, very long time and have never run into an issue caused by these coplanar `difference()` instances. Where should I be seeing the issues that this "feature" helps me solve?

UBaer21, 2022-06-23:
e.g. you generate two spheres via polyhedra  one as iso.sphere the other polar  and you want to see where they overlap and where not.   Or you put a cube on the corner and align with a cylinder($fn=3); manually.  In general you should not generate non-manifold geometry. But fixing those cause deviations. 
```
cube(10);
translate([10,10])cube(10);
```
  how would you fix this .. deviate the angles, or change the size - this could lead to cascading effects if you have a grid of cubes.. where you have to decide which needs to change.  That is why you design with overlaps.

UBaer21, 2022-06-23:
or you just import a rotated cube and now want to attach a cylinder orthogonal to its side.

UBaer21, 2022-06-23:
@DeflateAwning  you can read about this in https://en.wikibooks.org/wiki/OpenSCAD_Tutorial/Chapter_1
"two_cubes_barely_touching.scad"  
Another reason is that the resolution ($fs) can change so without overlap you may generate voids when putting a cylinder in a hole.

MichaelAtOz, 2022-06-23:
The problems often go unnoticed.

Apart from non curved, axis aligned objects, all the rest are approximations.

As eventually floating point gets involved you get tiny inaccuracies.

 

If left to chance, a point will either be, less likely, just at the sweet spot, or on the right or wrong side.

 

You may get the dreaded CGAL Error, or a warning, but the flaws produced don't always get detected.

The world is full of STL's with a 'degenerate face' or which are 'non-manifold'. (search for those terms here, I have debugged soo many)

It depends on what subsequently processes these STL's whether they are a minor or major problem.

Mostly Slicers don't get bothered, so lots go unnoticed, but upload it to Shapeways etc or import() it and you will find out. 

 

You need to engineer your geometry to make it clear whether one object is definitely embedded in another, or clearly separate.

 

This is the model which drove me insane (I'm better now, mostly) trying to debug

 

There are so many curves meeting at all sorts of angles.

You can't look at that and somehow decide which surface to nudge automatically, nudge one way and you bump into something else.

Then unless you've engineered it, all it takes is a $fs change and all those points move somewhere else.

 

 

 

  _____  

 

 

This email has been checked for viruses by AVG.
https://www.avg.com

MichaelAtOz, 2022-06-23:
Aargh, I forgot how bad GitHub is with emails...

Here is the model which drove me insane:
[image]