openscad/openscad #4287: Feature Request: allow centering cubes by axis

DeflateAwning, 2022-06-23:
I want to be able to center a cube on only some of its axes, and not all of them. I want to be able to specify which axes I want to center the cube on.

Specifically, I want the following OpenSCAD to be equivalent:

```
// how it currently must be done:
translate([-20/2, 0, -45/2]) cube([20, 10, 45]);

// how I want it to work
cube([20, 10, 45], center=[true, false, true]);
```

Another example taken out of a recent project, which shows how quickly the way it currently must be done gets unruly:
```
// current way:
translate([0, -.5*(antcap_d + screwplate_l*2), 0]) cube([screwplate_t, antcap_d + screwplate_l*2, screwplate_h]);

// new proposed way
cube([screwplate_t, antcap_d + screwplate_l*2, screwplate_h], center=[false, true, false]);
```

My OpenSCAD Version: 2021.01

t-paul, 2022-06-23:
Duplicate of #265.

DeflateAwning, 2022-06-23:
If this is completed, then what is the syntax to use this feature? Thanks!

t-paul, 2022-06-23:
It's not completed, the issue is closed because it's a duplicate of another (much) older still open issue.