openscad/openscad #4288: CGAL error when using `projection(cut=true)`

frazar, 2022-06-24:
When running the following code with OpenScad 2021.01-5:

```
projection(cut=true) translate([126.5,179,-170 + 65.1]) rotate([+90,0,-90]) import("F5.stl");
```

using [this stl file](https://drive.google.com/file/d/1qs-BTrjSA24O2R1n4Kr74Jv2HMXXJhEX/view?usp=sharing) as input I get the following error message:

```
 CGAL error in CGALUtils::project during bigbox intersection: CGAL ERROR: assertion violation!
Expr: target(s1).vertex()==target(s2).vertex()
File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 219
```

UBaer21, 2022-06-24:
seems to be related to this stl (or size/resolution), i got that error also on different positions.
But 
`projection(cut=true) translate([-150,-150,-50]) import("F5.stl");`
is working fine when only the ring is cut.
[image]

UBaer21, 2022-06-24:
when cutting the stl i find several voids
```
intersection(){
  translate([126.5,179,-170 + 65.1]) rotate([+90,0,-90])  import("F5.stl",convexity=10);
  cylinder(.0001,d=1000);
}
```

[image]

So i assume you need to repair the stl to get results.

frazar, 2022-06-25:
Ops! Sorry for mistakenly reporting this issue, I am not particularly proficient with .stl files.

If I'm not asking too much, could you give me some general indications as to how to repair the stl?

Thank you again for you help.

UBaer21, 2022-06-25:
[F5.tar.gz](https://github.com/openscad/openscad/files/8984535/F5.tar.gz)
 here the result (svg) and a re-meshed and reduced version.  There are some tools to repair geometry http://meshrepair.org/

frazar, 2022-06-27:
Amazing! Worked like a charm. Thank you very much

closed 2022-06-27 by frazar