openscad/openscad #4289: Collapsed console area

nfedyashev, 2022-06-25:
[image]

Linux(debian), i3
I'm not sure what changed because it worked fine last time I used it(same version).
I can't find a way to enlarge it after it collapsed.

The only workaround to find out what's in the console is right-click, select-all, copy and paste it elsewhere.

thehans, 2022-06-25:
You should be able to click and drag the "border" between the 3D view and Console.  There is a barely visible indicator of this draggable area.  

[image]

Although the dotted line is narrow and centered in this case, clicking anywhere along this horizontal should work.

Unfortunately, I think the visibility of this is dependent on your OS and desktop theme settings rather than something controlled directly by OpenSCAD.

nfedyashev, 2022-06-25:
@thehans thank you!

closed 2022-06-25 by nfedyashev