openscad/openscad #4291: suggestion: provide customizer support for builtin fonts

yeroca, 2022-07-01:
It would be nice that from the customizer the user can select fonts from those that are installed on the system.

I'm thinking that a special annotation such as:

`font = "Liberation Sans:style=Italic"; // {installed_fonts}`

As it stands, the programmer has to guess at what fonts are installed on the user's system.

UBaer21, 2022-07-01:
menu help ⇒ font list   is not enough?
[image]

t-paul, 2022-07-02:
This should be doable without breaking compatibility with Thingiverse Customizer. I'd prefer that we get started with a native annotation style, but this seems like a quick win that will not block any later updates.

yeroca, 2022-07-02:
For a notation keyword, is there an existing annotation style?  I was thinking later that a syntax like

`// #installed_fonts
`
might be a little easier to implement.

t-paul, 2022-07-02:
The current logic needs to follow: https://customizer.makerbot.com/docs
Native annotation does only exist in proposal form, syntax not finalized: https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases

Looking at the makerbot docs, using additional tags (that will be - hopefully - ignored on Thingiverse) with `{` `}`is maybe not a bad choice.

yeroca, 2022-07-02:
Ah, interesting, so eventually you might end up with something like
`
// @InstalledFonts()`

as a syntax, but it'd be a tail-wagging-the-dog situation if you went ahead with that at this point in time.

yahbluez, 2024-10-25:
@t-paul  thank you for moving me to this issue.
That is indeed the same issue.
I did not search back 2 years.

Is this really such a big problem, to find a way to make the list of installed fonts visibly for the common customizers?
 I like to motivate some kind of new customizer paradigm to make enhancement easier without breaking old stuff.

jordanbrown0, 2024-12-28:
Note that you can use Help / Font List / ... / Copy to Clipboard and then paste into a Customizer string field.  Not as classy and seamless as one might like, but a lot better than guessing.

yahbluez, 2024-12-28:
@jordanbrown0 yah, but workarounds are not a real solution. It would be nice if the openscad developer community would spend some mind on their users that uses openscad via the so called customizers.

Today i understood the problematic situation, that the customizer was somehow quick and dirty implemented by thingiverse and is now also used by makerworld.

Maybe that opens up the door to implement a kind of customizer API that allows websites to use a upgraded kind of customizer. I fully understand and accept that making openscad more usable is not one of the openscad developer goals but maybe this request open up the door a little bit to a more open mind. 

Things like export() or file() to process data are also on my private list of enhanced functionality. 
Stuff that needs additional python code today but could be done inside of openscad without breaking the security directive.

jordanbrown0, 2024-12-29:
I *would* like some kind of Customizer tag that would pop a font picker.  While thinking about that, I realized that the font-list trick would work, with only a couple of extra clicks.  (But you do have to delete the quotes too.)

yahbluez, 2024-12-29:
Good morning Jordan.

A customizer tag would open the door to have a customizer that is future
proof.

***@***.***("name tag", values,...)

That way a single pattern ***@***.***() offers everything the
implementation has to offer for the customizer
and the "name tag" opens up the door to any future change or enhancement
without harming older stuff.

Also doing this as a comment let the script still be backward compatible to
any older version of openscad.

To avoid that this again would be used in different ways
(thingivers/makerworld) the first step should be to describe any name tag
structure that is used today by the customizer.

If I think about it, this way may also allow the implementation of export()
without adding the export() to openscad main code which is not wanted from
openscad developers. In some projects we used tags to let python know how
to run and export. With a tag described here that could be done too.

The idea to use a single tag is a good idea.

best regards