openscad/openscad #4297: [Feature request] Allow touch-pad friendly 3D navigation

fancsali, 2022-07-12:
Using a touchpad, and navigating in the 3D view can be somewhat difficult: clicking the right button and moving around at the same time is a bit tricky, as right click is usually implemented as "keep two fingers on the pad and keep it depressed".

Most users, who use a _scripted CAD_ are probably approaching this from a programmers perspective anyway, and do not necessarily use a separate mouse, as they are more likely to be in the keyboard-first camp.
 
Hence, I am proposing using FreeCAD's already established approach for the purpose: holding down `Shift` and moving around with the mouse will **pan**, while keeping `Alt` depressed and moving around would **rotate** the view.

Still, for those amongst us, who do like to use the mosue: this is something that IMO works equally nice with a touchpad and an actual mouse and matches the already established habit with the _other big FOSS CAD package_.

Besides not reinventing the wheel, this also caters for anyone using both - which I expect is a considerable amount of fellow geeks.

fancsali, 2022-07-13:
Looking at the [code](https://github.com/openscad/openscad/blob/ea8950bd3cd7f06da333c15463492e484098e591/src/gui/QGLView.cc#L240), and the [documentation](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_User_Interface#View_navigation); and discussing this in the matrix channel, it turns out there are a few more aspects of this:

* Some (and in particular the `Shift`) modifier key(s) is/are already bound to functionality around manipulating the view.
* On Mac machines `Ctrl`-`Left mouse` is actually an alias of `Right mouse`
* There are many other thoughts and ideas around this topic, e.g. https://www.reddit.com/r/openscad/comments/vnegj2/change_mouse_view_setting/

After some discussion on the [matrix](https://app.element.io/#/room/#openscad:libera.chat) channel we've established, that:
* This could be achieved to various degrees on various OS's with mouse emulation for touch-pads (which might or might not be built into the OS/default driver)
* Most modern OSs support multi-touch, so left/right click two/three finger touch and click can be achieved - but that's not the focus here, it's rather a way of achieving this with modifiers rather multi-touch, if (due to the script-CAD nature) the user's hands are on the keyboard anyway.
* Apart from that, the way forward might be to allow the user to select which modifier (`Alt`, `Ctrl`, `Shift`, `Left Mouse`, `Right Mouse`, etc...) should trigger which feature (`Rotate`, `Rotate along axis towards user`, `Pan`, `Zoom`, etc.) - this however takes more effort.

fleiner, 2022-12-09:
This patch will change `<ctrl>` - left mouse to pan, and `<shift>-<ctrl>` left mouse to zoom, which solves my touchpad problem.

    diff --git a/src/gui/QGLView.cc b/src/gui/QGLView.cc
    index 93c4ebfab..af9ed7a67 100644
    --- a/src/gui/QGLView.cc
    +++ b/src/gui/QGLView.cc
    @@ -275,7 +275,7 @@ void QGLView::mouseMoveEvent(QMouseEvent *event)
       if (mouse_drag_active) {
         mouse_drag_moved = true;
         auto button_compare = this->mouseSwapButtons?Qt::RightButton : Qt::LeftButton;
    -    if (event->buttons() & button_compare
    +    if (event->buttons() & button_compare && !QApplication::keyboardModifiers().testFlag(Qt::ControlModifier)
     #ifdef Q_OS_MAC
             && !(event->modifiers() & Qt::MetaModifier)
     #endif

ncsaba, 2023-12-17:
Check out this possible solution done for cadquery:
https://github.com/CadQuery/CQ-editor/issues/419
Essentially it is possible to do panning instead of rotation after a former additional touch (just like a double touch ending in a drag) for touch-screens, and a kind of triple-touch ending in drag for touch-pads (not sure why the difference, but on my box that's how it works). It was surprisingly small patch, and works perfectly for me. Maybe something similar could be implemented for OpenSCAD, possibly enabled by a configuration option ? I'm more of a python guy than C, but if needed I can try to do it, if it has some chances to be accepted.
Cheers,
Csaba

t-paul, 2023-12-17:
Ideally we do want to have this configurable, but right now that seems far out, so having a small change sounds like a good option.

To have a chance, it should
* be a small change, reviewing huge changes is difficult and has a large backlog already
* not just change existing behavior, either have a config option or extend in some compatible way
* try using Qt features, e.g. is this not working via [mouseDoubleClickEvent](https://doc.qt.io/qt-5/qwidget.html#mouseDoubleClickEvent)?

ncsaba, 2023-12-23:
Hi gain - I had a look at the OpenSCAD source code, and a simple patch for allowing the "tap + short pause + tap-drag" combination to pan instead of rotate was the only thing I am able to produce. Anything else is too much for my lacking C++ skills and time. The patch in it's current form will likely not interfere with any normal operation, but I did not test it thoroughly enough to be confident about that. 
[touch_pan_mode.patch](https://github.com/openscad/openscad/files/13760419/touch_pan_mode.patch)
It should apply on the current head of the git repository.
From my POV that's it at the moment, for me it works now and I unfortunately don't have the necessary time to make it any nicer.
Cheers,
Csaba