openscad/openscad #4299: Segmentation fault while doing nothing in particular

eliasjonsson023, 2022-07-14:
Sometimes OpenSCAD crashes while windows not even being in focus and sometimes when I press CTRL + n.
`[e@zen ~]$ openscad 
Segmentation fault (core dumped)
[e@zen ~]$ 
`
Just wanted you devs to know.

`[e@zen ~]$ openscad -v
OpenSCAD version 2021.01
`

eliasjonsson023, 2022-07-15:
I got OpenSCAD via pacman so I guess there is only the latest version.
I run EndeavourOS. (What more is there to say about version)

I will run OpenSCAD in order to verify that the bug is repeatable.

rcolyer, 2022-07-15:
@eliasjonsson023 Can you attach the contents that you see under Help, Library Info?  It of course is not happening on all systems, so that info might give some hints about what's different with your particular system, especially if someone else also experiences it, and sees something in common.

eliasjonsson023, 2022-07-15:
Sure, here it comes!
`OpenSCAD Version: 2021.01
System information: Linux 5.18.10-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 07 Jul 2022 17:18:13 +0000 x86_64 EndeavourOS Linux () 8 CPUs 15.30 GB RAM
User Agent: OpenSCAD/2021.01 (Linux x86_64; EndeavourOS Linux ())
Compiler: GCC "12.1.0" 64bit
MinGW build: No
Debug build: Yes
Boost version: 1_79
Eigen version: 3.4.0
CGAL version, kernels: 5.4, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.5
QScintilla version: 2.13.3
InputDrivers: 
GLib version: 2.72.2
lodepng version: 20180910
libzip version: 1.8.0
fontconfig version: 2.14.0
freetype version: 2.12.1
harfbuzz version: 4.3.0 (runtime: 4.4.1)
cairo version: 1.17.6
lib3mf version: 1.8.1
Application Path: /usr/bin
Documents Path: /home/e/.local/share
User Documents Path: /home/e
Resource Path: /usr/share/openscad
User Library Path: /home/e/.local/share/OpenSCAD/libraries
User Config Path: /home/e/.config/OpenSCAD
Backup Path: /home/e/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/home/e/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/home/e/.local/share/fonts
/home/e/.local/share/flatpak/exports/share/fonts
/var/lib/flatpak/exports/share/fonts
/usr/local/share/fonts
/usr/share/fonts
/home/e/.fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/home/e/Library/Fonts
/usr/share/fonts/TTF
/usr/share/fonts/adobe-source-code-pro
/usr/share/fonts/adobe-source-han-sans
/usr/share/fonts/cantarell
/usr/share/fonts/encodings
/usr/share/fonts/gsfonts
/usr/share/fonts/liberation
/usr/share/fonts/noto
/usr/share/fonts/encodings/large
GLEW version: 2.2.0
OpenGL Version: 4.6 (Compatibility Profile) Mesa 22.1.3
GL Renderer: Mesa Intel(R) Iris(R) Plus Graphics (ICL GT2)
GL Vendor: Intel
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(0), stencil(0)

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
[... 275 lines cut ...]
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_shader_interlock
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays`

kittlerm, 2022-07-22:
I had same problem after upgrading distro. OpenSCAD run fine just til upgrade. Searching kernel message on the internet take me to https://gitlab.freedesktop.org/mesa/mesa/-/issues/3570 and the [@lannop](https://gitlab.freedesktop.org/lannop) answer works for me, just run:

`MESA_LOADER_DRIVER_OVERRIDE=i965 openscad`

eliasjonsson023, 2022-07-23:
The above command returned errors upon starting OpenSCAD.
`[e@zen ~]$ MESA_LOADER_DRIVER_OVERRIDE=i965 openscad`
`libGL error: MESA-LOADER: failed to open i965: /usr/lib/dri/i965_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/dri, suffix _dri)`
`libGL error: failed to load driver: i965`

 However, since i have an iris GPU, i ran
`ls /lib/dri`
and saw iris_dri.so so I just ran:
`e@zen ~]$ MESA_LOADER_DRIVER_OVERRIDE=iris openscad`
and it just worked. THANK you for the advice!

closed 2024-02-04 by t-paul