openscad/openscad #4301: Feature Request: Make OpenSCAD language more normal

johannespfister, 2022-07-19:
It would be very helpful if the OpenSCAD language would be more like any other programming language.
To be more specific, i really miss this features:

- Support for and, or, xor and shift operations #3345
- Mutable variables
- while loop support #1744
- Real functions, that can have multiple statements and one or more return statement.
- structs or dictionaries. #4203
- Hex, octal and binary literals
- Pointers
- Integers
- Enums

Yes i know that you can work around most of this limitations, but they are often a pain and hard to read. There are good reason the wast majority of languages have a direct support for this features.

MichaelAtOz, 2022-07-19:
Well there are very good reasons for Immutable variable. I would not support that change.

I fully support your effort to improve the code base by implementing any of your other suggestions.

nophead, 2022-07-19:
You can already have as many lines as you want in a function using let(). They just all have to be assignments.

Pointers: definitely no place for them in a modern high level language.

johannespfister, 2022-07-19:
> You can already have as many lines as you want in a function using let(). They just all have to be assignments.
> 

This doesn't work:

```
z=5;

b=9;
if(z==5)
  { let(b=4); }
else
  { let(b=2); }
cube([2,2,b]);
```

> Pointers: definitely no place for them in a modern high level language.

Almost every (modern) language has pointers. In many of them, such as Python, every variable is a pointer.

rcolyer, 2022-07-19:
On enums, structs, and dictionaries, I think most of us agree these would be good additions in some syntax form.  The object literals proposal has been out there for a while and would fill much of this need, it just isn't complete yet.

It's not clear what the use case is for integers and why there would be a utility for this not provided by the double type for the OpenSCAD target purposes.  Doubles are good for representing integer values across 53 bits, and there are not typically needs for integer values larger than 2^53 in this sort of model design.  We would expect no discernible runtime improvements from adding integers because the calculations of math on doubles have close to zero impact on the overall performance of scad programs.  So without a clear benefit it's hard to justify an increase in language complexity and learning complexity for dealing with and recognizing a different numeric type.  The explicit support for bitwise operators is the same and tied to the integers.  I had demonstrated in #3345 that these are basically just math operations and easy to have library support for in the rare cases they are useful.  (I'm willing to be shown wrong on the "rare cases" claim, but I currently think it's a fairly rare need.)  Is that a little slower?  Yes, technically.  But has anyone ever produced a naturally-produced (i.e. designed wisely for a real purpose, not for the point) example OpenSCAD design that performs discernibly slow because of the lack of built-in bitwise operators and the need to use math operations instead?

I think no one opposes hex, octal, or binary literals.  I suspect they have fairly low utility, but these have the nice benefit of being low maintenance and non-interfering, meaning they don't really introduce any complexity to the implementation, to the language, or any issues in learning it.  A clean syntax PR for those would probably be accepted fairly quickly.

We already have real functions with multiple statements.  I routinely write functions with 50+ statements, which are all composed of a sequence of function statements inside of the let clause.  Multiple return values are also possible and something I routinely do.  The syntax for clarity purposes puts these in one place with the ternary operator deciding between them, which is a reasonable design choice as it keeps the human parsing of the result fairly straightforward.  It's by design a declarative language, not an imperative language, so there is no role for a "return" other than the evaluated result, but you can certainly choose between any number of those ranging up to fairly long chains.

Mutables and while loops are in the same cluster, and generally opposed because OpenSCAD is intentionally a stateless declarative language.  The only thing that determines the output of any function or module call, or the entering of any block, is the state when it enters.  There are not supposed to be any magic changing values hiding inside of anything.  Now we do have C-style for loops in the list comprehension context, which provide essentially the same behavior as a while loop, but without violating the declarative nature.  What I think has wide support is expanding out the C-style for loops so that these also work in the geometry contexts.  These again preserve the declarative nature because the iterative updates happen in the third clause, and thus there are formally no mutable states within-block introduced, but simply value updates which depend upon the previous iteration.  I'm confident a clean PR introducing this behavior would get accepted, and provide the desired behavior without disrupting the advantages of the declarative approach.  (Until such a PR is created, people can obtain these behaviors for the geometry case using module recursion, which already provides the clean state boundaries.)

And I think the only entry I missed addressing there is pointers.  I would have to say a hard no on pointers of any sort like a C pointer.  This is a huge hazard that has no role in OpenSCAD, where one of the core goals is to make a language that's completely safe for community sharing of 3D model code with no risks to the receiving system.  Now probably the bulk of what you are thinking from pointers is the ability to build up things that refer to other things.  We already have assignable function literals, and we aim for assignable module literals as a desired feature.  This covers 90% of the positive things pointer support could provide but without the hazards.  For the other 10% combining that with a flexible structure like dictionaries (and lists that we already have) could provide support for fairly complicated data structures that refer to arbitrary things within the program, again without the hazards.

nophead, 2022-07-19:
Not sure where this comment went but:

This doesn't work:
```
z=5;

b=9;
if(z==5)
  { let(b=4); }
else
  { let(b=2); }
cube([2,2,b]);
```

No it doesn't but this does:
```
z=5;

b=9;
b2 = z == 5 ? 4 : 2;

cube([2,2,b2]);
```

t-paul, 2022-07-19:
This needs splitting out single topics that get some common support and have a chance to get implemented. I'll keep it open for the discussion, but it's not going to be a feature request as it's much too broad.

1) bit operations + 8) integers
I guess those are linked to really make sense. I'm not sure it's worth introducing a new data type, so I'm with @rcolyer in that. This may get complex too as there's currently no useful support for type conversion which probably would be needed too. I think this needs some good arguments with examples to show how it can be useful.
2) mutable variables
General mutable variables are unlikely to happen, there's ideas/proposals for some local variables (restricted to their scope) which could help implementing algorithms without tossing out the overall immutable nature.
3) while loop
I'm not sure if there's a reasonable combination with 2) but otherwise the linked issue already explains why it's unlikely to happen
4) multi line function
Should be possible, it might be mostly syntactic sugar around let() as mentioned above
5) structs
Yep, internal support already exists, mostly needs final specification of syntax and behavior
6) numbers with different bases
Might be difficult to find a syntax that does not clash with the odd variable naming, but if there's a good suggestion on that, it should be easy to implement
7) pointers
While OpenSCAD does eat memory for breakfast, I don't see any reason for introducing memory management into the language. What scenario would need pointers?
8) integers, see 1)
9) enums
Yes, enums might be mostly good to make code read easier, which is a good thing. That said, I wonder if we partially get those for free with the object syntax.

jordanbrown0, 2022-07-19:
This is a topic that comes up once or twice a year.  At a super high level, mostly the answer is that OpenSCAD is designed around a number of principles that are *not* used in other languages.  If you don't like those principles, then OpenSCAD is not the droid you're looking for.

It would be really cool if somebody were to build a 3D modeling IDE with a pluggable language, where you type source text into the IDE, the language engine processes it into a CSG tree, and then a geometry engine displays the result and exports it as desired.  But nobody has spent the effort to make that happen, and the OpenSCAD developers don't have any interest in working on it, and arguably when you were done it wouldn't properly be called OpenSCAD.  (Carsten Arnholm's AngelCAD comes close.  I threw together a prototype that ran Python.)

---

This particular "make it a normal language" list doesn't include file I/O, but that's another common request.  One huge caution there:  it's basically impossible to write malware in OpenSCAD.  It's hard to say that of any "normal" language.

---

I agree with the "hard no" on pointers per se.  One could imagine references (as found in Java, JavaScript, Python, et cetera), but in an immutable-variable environment I don't think it's possible to tell the difference between a reference and a copy.

---

Bitwise operators and non-decimal constants... shrug.  I don't see any important applications, but they don't bother me.  Since they're well-defined for 53-bit integers embedded in doubles, and since I see even fewer applications for >53-bit integers, I don't think they would need to be tied to support for integers.

Support for integers is one that I'd put a medium-hard "no" on.  I can't come up with any practical need, and they would introduce complexity and opportunity for error.

---

In my mind the most interesting part of enums is type safety, that you can't assign MONDAY to a variable of type "month".  But OpenSCAD doesn't have that kind of type safety, because it doesn't have declarations, and so I don't immediately see why `enum week { MONDAY = 1, TUESDAY = 2, ... }` would be an improvement over `MONDAY=1; TUESDAY=2; ...`.  (Maybe `echo` could emit the name rather than the number.)

Yes, object syntax would let you say `week = { MONDAY=1; TUESDAY=2; ...}` and then talk about `week.MONDAY`, but that's only slightly different from `week_MONDAY = 1;`.

---

One area where bitwise operators and enums come up is in managing sets of flags, but there I'd say that the right answer is in that word "sets".  Native support for sets would be better than telling people to encode sets as bit masks.

tkircher, 2022-07-23:
OpenSCAD is a functional style programming language, superior to imperative languages. If you find it hard to do what you are trying to do, it's probably because you're trying to do it wrong.

jordanbrown0, 2022-07-23:
It's probably not helpful to characterize one style as superior to the other, because opinions on the question differ.

What's clear is that they are different, and deliberately so, and OpenSCAD made imperative would not be OpenSCAD.

endrift, 2022-12-05:
Regarding some of these features, I do have some use for a few of these things, like bitwise ops (there are tricks you can do with them that are useful in some cases, and them not working is what led me to this thread), proper functions, mutable variables, and maybe some sort of structured types. However, the rest I can't see good arguments for:

- Pointers or even references have no use here without structured types, though pass-by-reference might be useful if there are structured types and functions. Otherwise there's really no use case at all.
- I can't really see a use case for binary or hexadecimal literals, but they're not that difficult on their own. I know I've been frustrated at times by languages like GLSL lacking them. Octal though? Does anyone use octal for anything other than file modes on Unix?
- I'm not sure what integers would be useful for here. While loops and stuff might have a use for integral subsets of doubles, double has a 51 bit mantissa so it's plenty enough for the vast majority of whatever you'd be doing with integers.

> OpenSCAD is a functional style programming language, superior to imperative languages. If you find it hard to do what you are trying to do, it's probably because you're trying to do it wrong.

Um, OpenSCAD is an _extremely_ imperative language. What do you think functional and imperative mean?

endrift, 2022-12-05:
Doesn't work; l, r, and b are only defined within the scope of each if/else block, not beyond them. That's why the mention of imperative was brought up at all.

nophead, 2022-12-05:
Rather than a low level construct like bitmasks, why not use a list of bools?
Also, what is wrong with this? It is shorter than if then else.

```
vars = some(x)&complicated(if_)>statement(with)-lots(of)%variables[foo]) ?
    a > 22 ? [14, 3, 7] :  [4, 5, 2]
 :  a < 13 ? [12, 2, 3]  : [10, 4, 1];
 
 l = vars.[0]; 
 r = vars.[1]; 
 b = vars.[2];
```

johannespfister, 2022-12-05:
> Rather than a low level construct like bitmasks, why not use a list of bools?

Because a list of bools is harder to read and harder to check.
You immediately see that bit 8 is set on `0x51E` but you have to count digits before you see if bit 8 is set on this `[ 0,1,1,1, 1,0,0,0, 1,0,1,0 ]` or this `[ 0,1,0,1, 0,0,0,1, 1,1,1,0 ]`. By the way, which is the correct order? The first one looks wrong because we normally use big-endian numbers in Latin based languages the other one is wrong because `array[bitposition]` doesn't work.
The other advantage of a simple integer is that you can compare it.
`versionText = bitmask==0x51E ? "Default" : "Custom"  ` How can i do that simply with a list of bools? 
Or how can i do something like this?
```
versionConfig= (bitmask<<3) | (someFlag<<2) | (someOtherFlag<<1) | (yetAnotherFlag<<0) ;
versionText = versionConfig==0x28F2 ? "Default" : versionConfig==0x28F0 ?  "Foo" : "Custom" ;
//In reality, i need a to check for more possible configurations than just 3.
```

The computer uses integers, it is a very natural object, easy to think about and easy to merge them (with or, and, nand, xor ... operations). A list is a relatively complicated structure, harder to merge and you need to consider edge cases such as out-of-bound access.

I now use a file with `0x0 = 0 ; 0x1 = 1 ; 0x2 = 2 ; ....` in order to be able to use hexliterals and use recursive functions (ugly but i try to stop thinking about it) and the pow() function for bit functions on floats. That is all dirty and only works up to 53 bits and then breaks without warning, but anyway.

> Also, what is wrong with this? It is shorter than if then else.

Thank you for your suggestion, probably the best way to solve it in OpenSCAD. But worse than with normal variables, it is harder to understand/read and it is easier to add/hide errors. Yes it is shorter but that is less important than readability and avoiding bugs.

Anyway, OpenSCAD is a great tool to for parameter based construction and supported well enough (FreeCAD ...) but from the point of view of a Programmer that is spoiled with easy to think about languages such as C and easy to use Languages such as Python, the language design of OpenSCAD is hard to use, hard think about, inefficient, cumbersome and far away from how the computer actually works.
The issue was created in the hope the OpenSCAD language can be a bit improved, at least the most annoying parts such as no support for variables.

tkircher, 2022-12-06:
> Hexadecimal is very useful when you work with Bitmasks. One example is is a casing for a PCB which has multiple connectors of the same type which may or may not be assembled. With a Bitmask i can specify which connector is placed and where there is a cutout in the case for that connector. A other example is working with Text. It is much faster to understand that the range of 0x20..0x7E is printable ASCII characters and needs less comments than why you check for the range of 32..126.

Using bitmasks in this way is a bad habit from C, and was only ever used along with logical operators because space and instructions were limited. Using modules you can specify which connector is placed and where a cutout would be, using a verbose name such as USB_connector_1. Also you're definitely the only person who thinks hexadecimal is clear. Even in C people use character literals because that is literate programming. Using magic hex numbers does not make code readable in any way.

nophead, 2022-12-06:
Yes bitmasks and shifting make sense writing device drivers and other low level software in C and C++ (which I spent a career doing) but I have never seen a need for it when describing 3D objects.

johannespfister, 2022-12-06:
@tkircher 
> Using modules you can specify which connector is placed and where a cutout would be, using a verbose name such as USB_connector_1.

Then you can use a GUI-CAD directly and don't need OpenSCAD. The main advantage of openscad is that you actually can use things like for-loops and place the same module multiple times without copying the module manually. In this example i have 4 connectors, all of the same type, that may or may not be placed. A single config variable to define which connectors are actually used and a for loop to check every one is much simpler than 4 modules and 4 ifs (which also violates DRY).

> Even in C people use character literals because that is literate programming.

Yes, when using as a single value. I can't really do that in OpenSCAD: `echo("c"+12);` reports an error.
By the way, to me `if(0x20<c&&c<0x7F) //check if c is printable` is more readable than ``if(' '=<c&&c<'~') //check if c is printable`.

tkircher, 2022-12-06:
@johannespfister I understand that for someone who is only used to C, it seems most natural, but many of us prefer domain-specific languages for specific purposes. In this case, OpenSCAD is not a programming language, it's a way of describing 3D models parametrically. It should definitely never be anything like C.

nophead, 2022-12-06:
`echo(ord("c") + 12);` works.

How many people that know hexadecimal don't know that 0x20 is 32 and 0x7F is 127?

So why is `if(c >= 32 and c <= 127)` less readable?

> versionText = bitmask==0x51E ? "Default" : "Custom" How can i do that simply with a list of bools?

`versionText = bitlist == [ 0,1,1,1, 1,0,0,0, 1,0,1,0 ] ? "Default" : "Custom";`

>versionConfig= (bitmask<<3) | (someFlag<<2) | (someOtherFlag<<1) | (yetAnotherFlag<<0) ;

`versionConfig= [yetAnotherFlag, someOtherFlag, someFlag, bitmask];`

jordanbrown0, 2022-12-06:
Note that ord(c) will return the numeric code point value for c, so you can say
```
if (32 < ord(c) && ord(c) < 127)
```

But really you should say
```
function isprint(c) = (32 < ord(c) && ord(c) < 127);
```
and thereafter say
```
if (isprint(c))
```

But also: there are about 149K printable characters outside that range, since OpenSCAD supports Unicode natively.

php4fan, 2023-10-14:
I got here via Google because, apparently, variables in OpenSCAD are immutable. By the way, I had never heard about the concept of "immutable variables" before (and I've been coding for 30 years). I think that's called a constant.

Anyway,  let's accept for a moment that variables are basically constants and there are no "mutable" variables (i.e. variable variables) and pretend that's ok.

Something I find really outrageous is that, if you try to assign a new value to a variable, you don't even get an error or a warning of any kind. It just silently ignores the assignment, leaving you wondering WTH is going on.

gsohler, 2023-10-14:
Hi phpfan, welcome to the club!

OpenSCAD is a wonderful tool to design and express 3D Objects, but at
the same time I also cannot understand either that Variables can only be
assigned once, because
this forces you to apply strange strategies (mostly recursive) to achieve
the same goal.

My personal solution to this issue is to use an additional language which
can express things easier - Python.
In case of interest, have a look to http://www.guenther-sohler.net/openscad
for details

JamesC1, 2024-04-14:
> we aim for assignable module literals

Would that let me do something like this?
```openscad
for (option = [
    [1,"union", union()],
    [2,"intersection", intersection()],
    [3,"rotate", rotate([90,0,0])],
    [4,"scale", scale([0.5, 0.5, 0.5])]
 ]) {
     echo(option[0], option[1]);
     translate([20*option[0], 0, 0])
     option[2]() {
         cylinder(h=10, r=5, center=true);
         sphere(10);
     }
 }
``` 

If so, yes, please I want it : - )

gsohler, 2024-04-14:
Are you looking for something like this ?

https://imgur.com/a/WB6t51G

from openscad import *

result=[]
for option in [
    [1,"union", union],
    [2,"intersection", intersection],
    [3,"rotate", rotate,[90,0,0]],
    [4,"scale", scale,[0.5, 0.5, 0.5]]
    ]:
    print(option, option[2])
    if len(option) == 4:
        item=option[2]( cylinder(h=10, r=5, center=True),option[3])
    else:
        item=option[2](
            sphere(10).left(8),cylinder(h=10, r=5, center=True)
        )
    result.append(item.right(20*option[0]))
show(result)

you are even getting  closer by compiling SCAD strings and evaluating them
by "scad"  function into a solid ...
cheers

JamesC1, 2024-04-14:
@gsohler, probably.  I haven't completed my first 3d designed in openSCAD yet, so I'm not yet ready to change tools.  By the time I am, could you edit the link to a good start point for that tool into your comment, please?

gsohler, 2024-04-14:
Hi James, no rush, you can test it  *any* time,

and even while testing you can still use SCAD language, use Python language
or even mix them in both directions.

A  brief tutorial which shows some basic steps can be found here:

http://pythonscad.org/tutorial/site/index.html

Have a nice day!

No rush

t-paul, 2024-04-14:
Ok, closing this as it's just way too broad in scope and some parts (e.g. Pointers) are not going to happen in any case. If someone want's to pick single topics and drive those on the requirements side, feel free to open a dedicated more focused issue.

closed 2024-04-14 by t-paul

Sod-Almighty, 2025-01-09:
> @endrift
> 
> It is much faster to understand that the range of 0x20..0x7E is printable ASCII characters and needs less comments than why you check for the range of 32..126.

Have to disagree. _Everyone_ knows that 32 is the first printable character and 127 is the last ASCII codepoint. 0x20 and 0x7E, on the other hand, have no meaning whatsoever to me.

Humans think in decimal.