openscad/openscad #4303: Proposal for how to center imported STL files

jpswensen, 2022-07-19:
I have seen some forum posts about people wanting the STL import to center the file, but I don't think it had been implemented yet. So, I went searching for an alternative solution. I found a python library called numpy-stl, which had an example of translating and rotating STL models. 

So, I made a simple modification to their code and found the min and max coordinates on each axis, found the centroid of all the points, and then translated the whole model accordingly. Here is that example code in python.

`
def find_mins_maxs(obj):
    minx = maxx = miny = maxy = minz = maxz = None
    for p in obj.points:
        # p contains (x, y, z)
        if minx is None:
            minx = p[stl.Dimension.X]
            maxx = p[stl.Dimension.X]
            miny = p[stl.Dimension.Y]
            maxy = p[stl.Dimension.Y]
            minz = p[stl.Dimension.Z]
            maxz = p[stl.Dimension.Z]
        else:
            maxx = max(p[stl.Dimension.X], maxx)
            minx = min(p[stl.Dimension.X], minx)
            maxy = max(p[stl.Dimension.Y], maxy)
            miny = min(p[stl.Dimension.Y], miny)
            maxz = max(p[stl.Dimension.Z], maxz)
            minz = min(p[stl.Dimension.Z], minz)
    return minx, maxx, miny, maxy, minz, maxz

def translate_single_axis(_solid, step, padding, multiplier, axis):
    if axis == 'x':
        items = [0, 3, 6]
    elif axis == 'y':
        items = [1, 4, 7]
    elif axis == 'z':
        items = [2, 5, 8]
    for p in _solid.points:
        # point items are ((x, y, z), (x, y, z), (x, y, z))
        for i in range(3):
            p[items[i]] += (step * multiplier) + (padding * multiplier)

def translate3D(_solid, step, padding, multiplier):
    itemsx = [0,3,6]
    itemsy = [1,4,7]
    itemsz = [2,5,8]

    for p in _solid.points:
        for I in range(3):
            p[itemsx[I]] += (step[0] * multiplier[0]) + (padding[0] * multiplier[0])
            p[itemsy[I]] += (step[1] * multiplier[1]) + (padding[1] * multiplier[1])
            p[itemsz[I]] += (step[2] * multiplier[2]) + (padding[2] * multiplier[2])

minx, maxx, miny, maxy, minz, maxz = find_mins_maxs(main_body)
w1 = maxx - minx
l1 = maxy - miny
h1 = maxz - minz

print(f'ORIGINAL minx:{minx}  maxx:{maxx}  miny:{miny}  maxy:{maxy}  minz:{minz}  maxz:{maxz}')

if l1 > w1:
    print('Rotating body so that long side is along the X axis')
    main_body.rotate([0.0, 0.0, 1.0], math.radians(90))

main_body.rotate([1.0, 0.0, 0.0], math.radians(180))

minx, maxx, miny, maxy, minz, maxz = find_mins_maxs(main_body)
w1 = maxx - minx
l1 = maxy - miny
h1 = maxz - minz

print(f'AFTER ROTATE minx:{minx}  maxx:{maxx}  miny:{miny}  maxy:{maxy}  minz:{minz}  maxz:{maxz}')

xc = (minx+maxx)/2.0
yc = (miny+maxy)/2.0

translate3D(main_body,[-xc,-yc,-minz], [0,0,0], [1,1,1])
minx, maxx, miny, maxy, minz, maxz = find_mins_maxs(main_body)
w1 = maxx - minx
l1 = maxy - miny
h1 = maxz - minz

print(f'AFTER TRANSLATE minx:{minx}  maxx:{maxx}  miny:{miny}  maxy:{maxy}  minz:{minz}  maxz:{maxz}')
main_body.save('MODEL_centered.stl', mode=stl.Mode.BINARY)
`

I then went and searched for where the import code for the STL files is located in the OpenSCAD codebase and found what I think are the two relevant locations:
https://github.com/openscad/openscad/blob/6eb6e59c7f6e70dde05970a5af90c5c88a41048d/src/core/ImportNode.cc
and
https://github.com/openscad/openscad/blob/3b6312b819d29df4329ec187b3f936ea3af21277/src/io/import_stl.cc

Am I wrong in thinking that I could add a center parameter to the import_stl() function and that at the end of that function before returning the Polyset variable "p", that I could perform this exact same process of finding the min/max on each axis and shifting the points in the Polyset accordingly?

t-paul, 2022-07-19:
That would be possible, but it's also very limited. I think a more flexible approach would be along the lines of the experimental `textmetrics()` which returns structured information about font/text. A similar solution should work for `import()`as well. Right now this kind of feature is specialized only for JSON files, but it's already in place to do:
```
data = import("file.json");
```

jpswensen, 2022-07-19:
So, just so I make sure I am understanding, you are thinking I should make this a standalone function like "modelmetrics" that could potential return the bounding box, the geometric center, the center of mass, _etc_. so that the user could use that information however they wish to translate as needed to either center, align an edge with an axis, _etc_. as the user sees fit?

That seems more flexible. I did a bit more inspection and it looks like none of the 3D importers handle the "center" parameters.

Is there a way to easily use a function like builtin_modelmetrics from other C++ code, so that if I did want to also implement the center parameter for the 3D importers that it would be easy to call that modelmetrics() function on the Polyset that is generated from the imports and then transform accordingly?

jordanbrown0, 2022-07-19:
Such a function might well return not only metrics, but the actual geometry data so that your OpenSCAD program could process it - e.g., to apply bends.

w1ebr, 2023-11-11:
Did  this ever get implemented?

jordanbrown0, 2023-11-11:
I don't believe so, but PR#4478 includes a more general solution that lets you extract all sorts of information about any geometric shape.

UBaer21, 2024-08-21:
would be great for SVG import if 
size=import("file.svg").size;
pos=import("file.svg").pos; 

could be used to get the bounding box or even possible ID, layer or labels (not accessible via import)

kintel, 2025-03-21:
#5671 Added a center= argument to import(), which also works for STL imports.