openscad/openscad #4311: Export color models

Rami5743, 2022-07-29:
It seems that until now, openscad does not allow to export of color models (though it does support presenting such models). 

I think this is a very important feature to add since 3d printing of color models become much cheaper now. 

Thanks a lot,
Rami

jordanbrown0, 2022-07-29:
See #1608.

I believe that a common mechanism for multi-color slicers is to give them one STL per color.  You would then correspondingly arrange for your OpenSCAD model to have a parameter that would let you generate each color individually.  Not nearly as convenient as built-in support, but do-able.

Rami5743, 2022-07-30:
Thank you very much.

I think that I do not really understand your answer. If my model has only one part, how can I use one stl per color? In many of my models, I have several "pieces" and each "piece" have its own color. However, the "pieces" are glued together, so they are not separate stl's. 

Thanks again,
Rami

jordanbrown0, 2022-07-31:
You would make one STL of the red parts, one STL of the green parts, et cetera. Then you import them all into your slicer with the same origin so they are reassembled correctly, and assign the desired colors for each.

kintel, 2024-07-19:
Fixed by #5185 - only OFF supported so far, but the feature is complete.

closed 2024-07-19 by kintel