openscad/openscad #4312: Exists in array / warning from search function

ygoe, 2022-07-30:
I'm in need of a way to determine whether a value exists in an array of arrays.

```js
wallMounts = [
	["y", 0],
	["-x", 10],
	["x", 10],
];

// .NET style:
if (wallMounts.any(m => m[0] == "y")) {
	// work here
}
```

Seems like the search function can be used, and it does work for me.
https://stackoverflow.com/a/35462281/143684

```js
echo(len(search("y", wallMounts)));
// prints 0, 1 or more, depending on the data
```

But it always generates a warning "WARNING: search term not found" when the value does not exist. Looks like that warning should be long removed, but I have it in version 2021.01 on Windows.
https://forum.openscad.org/Suppressing-quot-search-term-not-found-quot-warning-in-search-td12359.html

So, what to do here? Add another function or fix the search function?

UBaer21, 2022-07-30:
try 
`echo(is_num(search(["y"], wallMounts)[0]));`

ygoe, 2022-07-30:
That works better. The key point seems to be searching for `["y"]` instead of `"y"` (at the default array index of 0 in the data array). Judging from the examples 6 to 8 in the [documentation](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#search), this looks like an unintended side effect I can use here. I don't understand the description part, it's too complicated. This is a function with a million features. Should be split up.

PS: The link from the [cheatsheet](https://openscad.org/cheatsheet/index.html?version=2021.01) doesn't directly lead to the search function, I have to scroll down to it first.

UBaer21, 2022-07-30:
Thx for the hint .. links are fixed. 
And yes search is a bit complicated.
If simply searching a character in a string it works fine but for list the search value should be a list too else the string is used as individual searches "abc"  is looking for "a" "b" "c" while ["abc"] will look for "abc"  but will ["cat"] will not be found in "catdog"

jordanbrown0, 2025-03-08:
That message was removed in d33fc193968c12b3c3a159985e25bc67cb2b38ad.

closed 2025-03-08 by jordanbrown0