openscad/openscad #4313: Error when rendering imported STL: ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle()

kerrickstaley, 2022-07-30:
[(All files needed to repro this issue are available here.)](https://github.com/kerrickstaley/openscad-render-imported-stl-issue)

The following code shows a preview but fails to render for me:
```
union()
for (z = [0, 1])
translate([0, 0, z])
import("arduino-modified.stl");
```
It gives the error:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 418
```
The file `arduino-modified.stl` is a file that I exported from OpenSCAD itself.

Curiously, I *do not* get this error and the render works when I swap the .stl file for `Arduino.ipt.stl`, which was exported from Autodesk Inventor.

## System info
```
$ openscad --version
OpenSCAD version 2021.01
$ cat /etc/lsb-release 
DISTRIB_ID="Arch"
DISTRIB_RELEASE="rolling"
DISTRIB_DESCRIPTION="Arch Linux"
```

## Notes
I think this may be related to https://github.com/openscad/openscad/issues/3841, but I find that issue a little hard to read and I thought it was worthwhile to upload a clean repro.

UBaer21, 2022-07-30:
checked with DevSnapshot version (2022.05.04) and it renders without problem.
 (but  maybe better to use projection() not sure what you are trying to accomplish here) 
The modified.stl looks ok so far .. you can try if a different loop like [0,1.01] changes something you just hit a bad alignment causing problems.

MichaelAtOz, 2022-07-31:
The STL has holes & degenerate faces.
[image]
The holes are tiny, those orange lines point to two.
Shells is normally 1, I can't see any detached parts, it could be an artefact of the holes, or there may be hollows inside.
Other side
[image]

I also saw some strange things, but they were inconsistent, so probably other artefacts.

MichaelAtOz, 2022-07-31:
BTW `arduino.ipt.stl` is good.

nophead, 2022-07-31:
I was able to get the STL modified by OpenSCAD to be OK according to Netfabb by moving the cubes, so that they didn't have coincident faces, but I still got the same CGAL error when it was imported again and unioned with a cube.

MichaelAtOz, 2022-07-31:
> The file arduino-modified.stl is a file that I exported from OpenSCAD itself.

It appears to be the `arduino.ipt.stl`with the USB-B hardware removed, presumably in OpenSCAD.
The site I highlighted is the solder tabs.
Perhaps if you share the .scad code you used to remove it?

UBaer21, 2022-07-31:
@MichaelAtOz he did  here https://github.com/kerrickstaley/openscad-render-imported-stl-issue/blob/main/arduino-modified.scad  
It is a flush cut on the solder side, while to top is a 0.004 recess.

nophead, 2022-07-31:
It isn't uncommon for OpenSCAD not to be able to import STL files it has exported. Vertices too close together will do it.

kintel, 2024-04-01:
Status as of 2024-03-24: Works with Manifold, still fails with CGAL