openscad/openscad #4314: Feature Request: Save a Copy menu option

jbinvnt, 2022-07-31:
Add a new File menu option titled "Save a Copy." This option exists in other software such as LibreOffice, and its behavior is described [here](https://askubuntu.com/questions/398969/what-is-the-difference-between-save-as-and-save-a-copy). This would be helpful for when the user wants to save a benchmark copy or alternate version with a different name, but then keep making modifications to the original file without needing to reopen it.

Most users would likely expect to see the option placed below "Save As," which is where it is located in LibreOffice et al.

MichaelAtOz, 2022-10-02:
I just added a US$25 bounty for  #4314, expires in 1 month, **easy beer money**. 
(testing the Boutysource interface tool, but go for it if anyone wants it)

MichaelAtOz, 2022-10-05:
OK again, this one got the bounty treatment without me triggering anything.
That looks promising.

closed 2022-10-22 by t-paul

MichaelAtOz, 2022-10-23:
@RayyanAnsari you should claim the bounty.

RayyanAnsari, 2022-10-23:
@MichaelAtOz I already have claimed the bounty - I think you need to approve it.

MichaelAtOz, 2022-10-24:
OK accepted on BS, thanks.

I recall in the past getting an email from BS asking for acceptance. 
I didn't get one this time.
@t-paul is that how you remember it working?

PR #4392 (2022-10-22): Add "Save a Copy" menu option
Closes #4314
commit cd9543b1 (2022-10-22): Add "Save a Copy" menu option

Closes #4314