openscad/openscad #4316: ASCII STL export is slow

jordanbrown0, 2022-08-02:
For `sphere(10, $fn=100)`, binary STL export using
```
openscad -o /tmp/a.stl --export-format=binstl /tmp/a.scad
```
takes about 0.3 seconds, but ASCII STL export takes about 5.5 seconds.  (That's for the entire execution, as measured with "time" on my Windows system under MSYS2.)

That seemed like a lot, so I did some experimentation and ... it *is* a lot.

The time is almost all spent in the [`fromString` function](https://github.com/openscad/openscad/blob/ea8950bd3cd7f06da333c15463492e484098e591/src/io/export_stl.cc#L49).

I don't know exactly what's involved in the fancy C++ stuff that it's doing, but when I replaced it with a function that did the same thing with primitive operations, the time dropped to about 0.8 seconds, and the resulting file was bit-for-bit identical.

```
#include <cstdlib>
#include <cstring>
...
Vector3d fromString(const std::string& vertexString)
{
  double p0, p1, p2;
  const char *s = vertexString.c_str();
  p0 = atof(s);
  s = strchr(s, ' ') + 1;
  p1 = atof(s);
  s = strchr(s, ' ') + 1;
  p2 = atof(s);
  return Vector3d(p0, p1, p2);
}
```

Note:  yes, that's a horrible implementation if you don't know exactly what the string looks like.  But here we *do* know what the string looks like, and I just wanted to see how fast it would be.  Safer implementations left as an exercise for the reader.

But:  Will anybody care?  Binary STL is faster and smaller; why would you use ASCII STL?  And yes, it's slower than it needs to be, but will a few seconds really matter after you've gone through a multi-minute render?

You can make it a bit faster still by eliminating some of the gyrations that it goes through to do processing with ASCII strings rather than with floats or doubles.  An experiment says that drops it to 0.6 seconds, and the source is a lot simpler.  But then many (all?) of the normals are off in the fourth or fifth significant figure, compared to the original.  Does that matter?  Don't know.  Which one is more correct?  Don't know.  Theoretically it might also generate degenerate triangles, when vertices differ when represented as binary but become the same when represented as ASCII.

thehans, 2022-08-02:
The original reason for "processing with ASCII strings rather than with floats or doubles" can be seen here: #1853

jordanbrown0, 2022-08-02:
I'll agree with the "seems dubious" opinions there.  Decimal representations of binary floating point numbers are rarely precise.  Binary floating point representations of decimal fractions are rarely precise.  Converting back and forth is information-losing.  But there's value to not rocking the boat.

jordanbrown0, 2022-08-14:
performance
(So that the next time I search for "ASCII STL performance" I'll find this issue.)

pca006132, 2023-03-21:
I think the main problem is each `to_string` call requires an allocation, slower the thing down.
https://github.com/openscad/openscad/blob/c4de8520c17b065d51e416614c3876c6277be223/src/io/export_stl.cc#L107-L141

If you look at the code, it will perform N small allocations where N is the number of vertices, and allocation is slow... I think refactoring this into a loop instead of using transform can improve performance quite a bit, but I don't have benchmark to look at so I am not sure. Also, if we know the precision beforehand we can probably preallocate the buffer and further reduce the allocation calls required.

jordanbrown0, 2023-03-21:
It seems entirely plausible that the `toString` calls are significant too, but note that my experiments showed that `fromString` accounted for more than 80% of the time.

pca006132, 2023-03-21:
Oh, wow, I didn't see that it is a lambda that is called over and over again. Yeah toStrings are probably not that slow (the array is not that large) and the main problem is fromString.

ochafik, 2023-05-20:
@pca006132 after thorough tinkering (see https://github.com/openscad/openscad/pull/4643), I managed to bring down the cost of the toString / fromString routine by using an indexed mesh (which tesselate_faces was creating under the hood anyway), so each vertex is converted only once. Also the conversion uses strtof (faster than istringstream). Between that and removing some superfluous checks (tesselation already eliminates polys w/ duplicate vertices), some models get a 2x speedup in binary stl and a 3x speedup in ascii stl export time, although for most models the export is probably not the slowest bit.

> lambda that is called over and over again

That's a red herring, I did test this against inlined alternatives extensively when I introduced this lambda but should have indeed left a comment around ;-)