openscad/openscad #4317: Feature Request: Stl File Name

F14Claude, 2022-08-02:
It would be usefull to have the possibility to change programatically the suggested stl file name.
Something like 
	stloutput = "foo";
instead of the .scad file name.
Example:
the same file begins by an option for selecting two parts of an object and I want 
the stl file have two different names without retyping them each time.
I know there are other ways to do this (2 source file and a library) but not suitable
for me.

UBaer21, 2022-08-02:
Maybe using the command line for that could help.
 https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment

F14Claude, 2022-08-02:
Would be a last resort but uncomfortable.

t-paul, 2022-08-02:
There will be no imperative call for naming and/or writing files. A single assignment would also collide with other enhancement requests asking for export of multiple parts to separate files.

The solution needs to be declarative to follow the OpenSCAD behavior, some ideas collected at https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases

F14Claude, 2022-08-02:
I understand that this apparently procedural action would countervail OpenSCAD descriptional model.
My wish is not to issue an Export operation when the order is reached but only to set the default file name as we do with variable assignation.

t-paul, 2022-08-02:
The intent is clear, I'm just trying to clarify that variable assignment is problematic for various reasons and there's a different implementation strategy which scales for multi-file export and keeps the declarative nature.

Something like:
```
module part1() {
  cube();
}

module part2() {
  sphere();
}

@part(name = "the-cube")
part1();

@part(name = "small-sphere")
part2();
```

nophead, 2022-08-02:
I simply name any modules that make stls something_stl() and then have a Python script that exports them all using the command line.

F14Claude, 2022-08-02:
I do similar things with a C# program specially when the same object has to be produced with several different lenghts.

jordanbrown0, 2022-08-02:
This is in some ways related to the "volumetric color" problem, where you want to track the color of volumes (not faces) and export (among other things) multiple STLs to form a multi-color model.

jordanbrown0, 2022-08-02:
For instance, one could imagine an export mode that was "export separate models for each color, appending the color name to this base name", or another that was "export only <color> parts of the model".  "Color" could of course be red, blue, green, but it could also be PLA, PVA (water-soluble filament), TPU (flexible filament), or subcomponent-A, subcomponent-B.  The word "color" would no longer be appropriate, but the concept could cover several use cases.

(Dozens of details to be worked out, of course, like how do you say "those parts of subcomponent A that are supposed to be built in red TPU".)

jordanbrown0, 2023-11-18:
Duplicate of #3404.

jordanbrown0, 2023-11-18:
Sorry, incorrect close code.  Should have been "closed not planned", which includes "duplicate".

closed 2023-11-18 by jordanbrown0