openscad/openscad #4318: Feature Request: Stl File Correction

F14Claude, 2022-08-02:
Most of stl files produced by OpenSCAD contain defaults that can be corrected even if not blocking for printing.
Why OpenSCAD doesn't offer a post correction of stl files?
Something like "Free Online stl repair" from FormWare.
Very efficient and usefull.

t-paul, 2022-08-02:
What defaults? Please explain what this is issue about.

t-paul, 2022-08-02:
STL is not an ideal format for multiple reasons, OpenSCAD has similar issue importing STL. I don't think there's much that can be done except maybe looking at specific examples. It might be worthwhile looking at 3MF format which can explicitely store geometry.
If there's a small example showing the problem, please create a dedicated issue with scad and stl files attached, I'll close this one as "fix any potential issue with STL" is basically impossible.

closed 2022-08-02 by t-paul

F14Claude, 2022-08-02:
Ok. I'll try to find a good and simple example.
For the rest I'll stick with using FormWare stl Repair.